ARMmbed/mbed-tools #55: Missing `MBED_BOOT_STACK_SIZE` in `MBED_CONFIG_DEFINITIONS`

hugueskamba, 2020-08-17:
### Description
The `MBED_BOOT_STACK_SIZE` macro is supposed to be created by the tools from [mbed-os/targets/targets.json](https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L59).

Please note that its value gets overridden by [the rtos configuration file](https://github.com/ARMmbed/mbed-os/blob/master/rtos/source/TARGET_CORTEX/mbed_lib.json#L65).

This issue is in part responsible for failure to build `NRF52840_DK` with CMake using the ARM toolchain

### Issue request type

- [ ] Enhancement
- [x] Bug

hugueskamba, 2020-08-17:
Is it what's expected in some scatter files? https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/device/TOOLCHAIN_ARM_STD/nRF52840.sct#L17

hugueskamba, 2020-08-17:
I also found this: https://github.com/ARMmbed/mbed-os/issues/10319
Where do we expect that macro to come from? And why does it build with the old tool?

rwalton-arm, 2020-08-17:
`MBED_BOOT_STACK_SIZE` is hard coded in the old tools. There’s a check for `target.boot_stack_size` and then it creates `MBED_BOOT_STACK_SIZE` in Python https://github.com/ARMmbed/mbed-os/blob/efe46b5aa17172374c0b40798a025ba6d9101274/tools/toolchains/mbed_toolchain.py#L969 - it's also duplicated in `mbed_config.h` as `MBED_CONF_TARGET_BOOT_STACK_SIZE`. 

In the new tool we're processing it from `targets.json` but we are only creating `MBED_CONF_TARGET_BOOT_STACK_SIZE`. This is really a problem with the configuration system in that we're producing duplicate compile time definitions with slightly different names (which are referenced using both names in different places throughout mbed-os). It's a similar story for `XIP_ENABLE` and `TFM_LVL`. 

The proper fix here would be to only define this once (probably using the `MBED_CONF_TARGET_BOOT_STACK_SIZE` name), and change the scatter files to use that macro to avoid this weird duplication. For now we will patch it in CMake to give the linker script maintainers time to make the transition.

hugueskamba, 2020-08-18:
Looks like this will be fixed in Mbed OS with: https://github.com/ARMmbed/mbed-os/pull/13452

Patater, 2020-10-28:
Closing, as ARMmbed/mbed-os#13452 has been merged.

closed 2020-10-28 by Patater