ARMmbed/mbed-tools #64: Unable to compile on Windows due to command length limits

nat45928, 2020-08-24:
### Description

While testing the new CMake tooling on a Windows machine I was unable to successfully compile the blinky example for the K64F target. I was able to configure CMake successfully but when building ninja output a wall of text outputting details of failed compiled commands was observed.

For example, the output of `cmake --build` generated a few instances of this:

```
[18/846] Building CXX object mbed-os/CMakeFiles/mbed-os....nectivity/drivers/cellular/UBLOX/N2XX/UBLOX_N2XX.cpp.obj
FAILED: mbed-os/CMakeFiles/mbed-os.dir/connectivity/drivers/cellular/UBLOX/N2XX/UBLOX_N2XX.cpp.obj
C:\PROGRA~2\GNUARM~1\92020-~1\bin\AR10B2~1.EXE -DARM_MATH_CM4 -DCOMPONENT_FLASHIAP=1 [...] -c ../mbed-os/connectivity/drivers/cellular/UBLOX/N2XX/UBLOX_N2XX.cpp
AR10B2~1.EXE: fatal error: cannot execute 'c:/progra~2/gnuarm~1/92020-~1/bin/../lib/gcc/arm-none-eabi/9.3.1/cc1plus.exe': CreateProcess: No such file or directory
compilation terminated.
ninja: build stopped: subcommand failed.
```

Where `[...]` is more compiler options (note the odd paths with ~ are short paths). In an attempt to debug this error I ran an example command directly in the commandline:

```
> C:\PROGRA~2\GNUARM~1\92020-~1\bin\AR10B2~1.EXE -DARM_MATH_CM4 -DCOMPONENT_FLASHIAP=1 -DCOMPONENT_SD=1 -DCPU_MK64FN1M0VMD12 -DDEVICE_ANALOGIN=1 -DDEVICE_ANALOGOUT=1 -DDEVICE_CRC=1 -DDEVICE_EMAC=1 -DDEVICE_FLASH=1 -DDEVICE_I2C=1 -DDEVICE_I2CSLAVE=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_LPTICKER=1 -DDEVICE_PORTIN=1 -DDEVICE_PORTINOUT=1 -DDEVICE_PORTOUT=1 -DDEVICE_PWMOUT=1 -DDEVICE_RESET_REASON=1 -DDEVICE_RTC=1 -DDEVICE_SERIAL=1 -DDEVICE_SERIAL_ASYNCH=1 -DDEVICE_SERIAL_FC=1 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_TRNG=1 -DDEVICE_USBDEVICE=1 -DDEVICE_USTICKER=1 -DDEVICE_WATCHDOG=1 -DFEATURE_PSA=1 -DFSL_RTOS_MBED -DMBEDTLS_CIPHER_MODE_CTR -DMBED_CONF_ALT1250_PPP_BAUDRATE=115200 -DMBED_CONF_ALT1250_PPP_PROVIDE_DEFAULT=0 -DMBED_CONF_ATMEL_RF_ASSUME_SPACED_SPI=0 -DMBED_CONF_ATMEL_RF_FULL_SPI_SPEED=7500000 -DMBED_CONF_ATMEL_RF_FULL_SPI_SPEED_BYTE_SPACING=250 -DMBED_CONF_ATMEL_RF_IRQ_THREAD_STACK_SIZE=1024 -DMBED_CONF_ATMEL_RF_LOW_SPI_SPEED=3750000 -DMBED_CONF_ATMEL_RF_PROVIDE_DEFAULT=0 -DMBED_CONF_ATMEL_RF_USE_SPI_SPACING_API=0 -DMBED_CONF_CELLULAR_CONTROL_PLANE_OPT=0 -DMBED_CONF_CELLULAR_DEBUG_AT=0 -DMBED_CONF_CELLULAR_MAX_CP_DATA_RECV_LEN=1358 -DMBED_CONF_CELLULAR_RANDOM_MAX_START_DELAY=0 -DMBED_CONF_CELLULAR_USE_APN_LOOKUP=0 -DMBED_CONF_CELLULAR_USE_SMS=0 -DMBED_CONF_DRIVERS_QSPI_CSN=QSPI_FLASH1_CSN -DMBED_CONF_DRIVERS_QSPI_IO0=QSPI_FLASH1_IO0 -DMBED_CONF_DRIVERS_QSPI_IO1=QSPI_FLASH1_IO1 -DMBED_CONF_DRIVERS_QSPI_IO2=QSPI_FLASH1_IO2 -DMBED_CONF_DRIVERS_QSPI_IO3=QSPI_FLASH1_IO3 -DMBED_CONF_DRIVERS_QSPI_SCK=QSPI_FLASH1_SCK -DMBED_CONF_DRIVERS_UART_SERIAL_RXBUF_SIZE=256 -DMBED_CONF_DRIVERS_UART_SERIAL_TXBUF_SIZE=256 -DMBED_CONF_ESP8266_BUILT_IN_DNS=0 -DMBED_CONF_ESP8266_DEBUG=0 -DMBED_CONF_ESP8266_POWER_OFF_TIME_MS=3 -DMBED_CONF_ESP8266_POWER_ON_POLARITY=0 -DMBED_CONF_ESP8266_POWER_ON_TIME_MS=3 -DMBED_CONF_ESP8266_PROVIDE_DEFAULT=0 -DMBED_CONF_ESP8266_SERIAL_BAUDRATE=115200 -DMBED_CONF_ESP8266_SNTP_ENABLE=0 -DMBED_CONF_ESP8266_SNTP_SERVER0=\"\" -DMBED_CONF_ESP8266_SNTP_SERVER1=\"\" -DMBED_CONF_ESP8266_SNTP_SERVER2=\"\" -DMBED_CONF_ESP8266_SNTP_TIMEZONE=0 -DMBED_CONF_ESP8266_SOCKET_BUFSIZE=8192 -DMBED_CONF_EVENTS_PRESENT=1 -DMBED_CONF_EVENTS_SHARED_DISPATCH_FROM_APPLICATION=0 -DMBED_CONF_EVENTS_SHARED_EVENTSIZE=768 -DMBED_CONF_EVENTS_SHARED_HIGHPRIO_EVENTSIZE=256 -DMBED_CONF_EVENTS_SHARED_HIGHPRIO_STACKSIZE=1024 -DMBED_CONF_EVENTS_SHARED_STACKSIZE=2048 -DMBED_CONF_EVENTS_USE_LOWPOWER_TIMER_TICKER=0 -DMBED_CONF_FAT_CHAN_FFS_DBG=0 -DMBED_CONF_FAT_CHAN_FF_CODE_PAGE=437 -DMBED_CONF_FAT_CHAN_FF_FS_EXFAT=0 -DMBED_CONF_FAT_CHAN_FF_FS_HEAPBUF=1 -DMBED_CONF_FAT_CHAN_FF_FS_LOCK=0 -DMBED_CONF_FAT_CHAN_FF_FS_MINIMIZE=0 -DMBED_CONF_FAT_CHAN_FF_FS_NOFSINFO=0 -DMBED_CONF_FAT_CHAN_FF_FS_NORTC=0 -DMBED_CONF_FAT_CHAN_FF_FS_READONLY=0 -DMBED_CONF_FAT_CHAN_FF_FS_REENTRANT=0 -DMBED_CONF_FAT_CHAN_FF_FS_RPATH=1 -DMBED_CONF_FAT_CHAN_FF_FS_TIMEOUT=1000 -DMBED_CONF_FAT_CHAN_FF_FS_TINY=1 -DMBED_CONF_FAT_CHAN_FF_LFN_BUF=255 -DMBED_CONF_FAT_CHAN_FF_LFN_UNICODE=0 -DMBED_CONF_FAT_CHAN_FF_MAX_LFN=255 -DMBED_CONF_FAT_CHAN_FF_MAX_SS=4096 -DMBED_CONF_FAT_CHAN_FF_MIN_SS=512 -DMBED_CONF_FAT_CHAN_FF_MULTI_PARTITION=0 -DMBED_CONF_FAT_CHAN_FF_NORTC_MDAY=1 -DMBED_CONF_FAT_CHAN_FF_NORTC_MON=1 -DMBED_CONF_FAT_CHAN_FF_NORTC_YEAR=2017 -DMBED_CONF_FAT_CHAN_FF_SFN_BUF=12 -DMBED_CONF_FAT_CHAN_FF_STRF_ENCODE=3 -DMBED_CONF_FAT_CHAN_FF_STR_VOLUME_ID=0 -DMBED_CONF_FAT_CHAN_FF_SYNC_T=HANDLE -DMBED_CONF_FAT_CHAN_FF_USE_CHMOD=0 -DMBED_CONF_FAT_CHAN_FF_USE_EXPAND=0 -DMBED_CONF_FAT_CHAN_FF_USE_FASTSEEK=0 -DMBED_CONF_FAT_CHAN_FF_USE_FIND=0 -DMBED_CONF_FAT_CHAN_FF_USE_FORWARD=0 -DMBED_CONF_FAT_CHAN_FF_USE_LABEL=0 -DMBED_CONF_FAT_CHAN_FF_USE_LFN=3 -DMBED_CONF_FAT_CHAN_FF_USE_MKFS=1 -DMBED_CONF_FAT_CHAN_FF_USE_STRFUNC=0 -DMBED_CONF_FAT_CHAN_FF_USE_TRIM=1 -DMBED_CONF_FAT_CHAN_FF_VOLUMES=4 -DMBED_CONF_FAT_CHAN_FF_VOLUME_STRS=\"RAM\",\"NAND\",\"CF\",\"SD\",\"SD2\",\"USB\",\"USB2\",\"USB3\" -DMBED_CONF_FAT_CHAN_FLUSH_ON_NEW_CLUSTER=0 -DMBED_CONF_FAT_CHAN_FLUSH_ON_NEW_SECTOR=1 -DMBED_CONF_FILESYSTEM_PRESENT=1 -DMBED_CONF_FLASHIAP_BLOCK_DEVICE_BASE_ADDRESS=0xFFFFFFFF -DMBED_CONF_FLASHIAP_BLOCK_DEVICE_SIZE=0 -DMBED_CONF_GEMALTO_CINTERION_BAUDRATE=115200 -DMBED_CONF_GEMALTO_CINTERION_PROVIDE_DEFAULT=0 -DMBED_CONF_GENERIC_AT3GPP_BAUDRATE=115200 -DMBED_CONF_GENERIC_AT3GPP_PROVIDE_DEFAULT=0 -DMBED_CONF_KINETIS_EMAC_RX_RING_LEN=2 -DMBED_CONF_KINETIS_EMAC_TX_RING_LEN=1 -DMBED_CONF_LORA_ADR_ON=1 -DMBED_CONF_LORA_APPLICATION_EUI="{0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}" -DMBED_CONF_LORA_APPLICATION_KEY="{0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}" -DMBED_CONF_LORA_APPSKEY="{0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}" -DMBED_CONF_LORA_APP_PORT=15 -DMBED_CONF_LORA_AUTOMATIC_UPLINK_MESSAGE=1 -DMBED_CONF_LORA_DEVICE_ADDRESS=0x00000000 -DMBED_CONF_LORA_DEVICE_EUI="{0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}" -DMBED_CONF_LORA_DOWNLINK_PREAMBLE_LENGTH=5 -DMBED_CONF_LORA_DUTY_CYCLE_ON=1 -DMBED_CONF_LORA_DUTY_CYCLE_ON_JOIN=1 -DMBED_CONF_LORA_FSB_MASK="{0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0x00FF}" -DMBED_CONF_LORA_FSB_MASK_CHINA="{0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF}" -DMBED_CONF_LORA_LBT_ON=0 -DMBED_CONF_LORA_MAX_SYS_RX_ERROR=5 -DMBED_CONF_LORA_NB_TRIALS=12 -DMBED_CONF_LORA_NWKSKEY="{0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}" -DMBED_CONF_LORA_OVER_THE_AIR_ACTIVATION=1 -DMBED_CONF_LORA_PHY=EU868 -DMBED_CONF_LORA_PUBLIC_NETWORK=1 -DMBED_CONF_LORA_TX_MAX_SIZE=64 -DMBED_CONF_LORA_UPLINK_PREAMBLE_LENGTH=8 -DMBED_CONF_LORA_WAKEUP_TIME=5 -DMBED_CONF_LWIP_ADDR_TIMEOUT=5 -DMBED_CONF_LWIP_ADDR_TIMEOUT_MODE=1 -DMBED_CONF_LWIP_DEBUG_ENABLED=0 -DMBED_CONF_LWIP_DEFAULT_THREAD_STACKSIZE=512 -DMBED_CONF_LWIP_DHCP_TIMEOUT=60 -DMBED_CONF_LWIP_ENABLE_PPP_TRACE=0 -DMBED_CONF_LWIP_ETHERNET_ENABLED=1 -DMBED_CONF_LWIP_IPV4_ENABLED=1 -DMBED_CONF_LWIP_IPV6_ENABLED=0 -DMBED_CONF_LWIP_IP_VER_PREF=4 -DMBED_CONF_LWIP_L3IP_ENABLED=0 -DMBED_CONF_LWIP_MBOX_SIZE=8 -DMBED_CONF_LWIP_MEMP_NUM_TCPIP_MSG_INPKT=8 -DMBED_CONF_LWIP_MEMP_NUM_TCP_SEG=16 -DMBED_CONF_LWIP_MEM_SIZE=33270 -DMBED_CONF_LWIP_NUM_NETBUF=8 -DMBED_CONF_LWIP_NUM_PBUF=8 -DMBED_CONF_LWIP_PBUF_POOL_SIZE=5 -DMBED_CONF_LWIP_PPP_ENABLED=0 -DMBED_CONF_LWIP_PPP_IPV4_ENABLED=0 -DMBED_CONF_LWIP_PPP_IPV6_ENABLED=0 -DMBED_CONF_LWIP_PPP_THREAD_STACKSIZE=768 -DMBED_CONF_LWIP_PRESENT=1 -DMBED_CONF_LWIP_RAW_SOCKET_ENABLED=0 -DMBED_CONF_LWIP_SOCKET_MAX=4 -DMBED_CONF_LWIP_TCPIP_THREAD_PRIORITY=osPriorityNormal -DMBED_CONF_LWIP_TCPIP_THREAD_STACKSIZE=1200 -DMBED_CONF_LWIP_TCP_CLOSE_TIMEOUT=1000 -DMBED_CONF_LWIP_TCP_ENABLED=1 -DMBED_CONF_LWIP_TCP_MAXRTX=6 -DMBED_CONF_LWIP_TCP_MSS=536 -DMBED_CONF_LWIP_TCP_SERVER_MAX=4 -DMBED_CONF_LWIP_TCP_SND_BUF="(2 * TCP_MSS)" -DMBED_CONF_LWIP_TCP_SOCKET_MAX=4 -DMBED_CONF_LWIP_TCP_SYNMAXRTX=6 -DMBED_CONF_LWIP_TCP_WND="(4 * TCP_MSS)" -DMBED_CONF_LWIP_UDP_SOCKET_MAX=4 -DMBED_CONF_LWIP_USE_MBED_TRACE=0 -DMBED_CONF_MBED_MESH_API_6LOWPAN_ND_CHANNEL=0 -DMBED_CONF_MBED_MESH_API_6LOWPAN_ND_CHANNEL_MASK=0x7fff800 -DMBED_CONF_MBED_MESH_API_6LOWPAN_ND_CHANNEL_PAGE=0 -DMBED_CONF_MBED_MESH_API_6LOWPAN_ND_DEVICE_TYPE=NET_6LOWPAN_ROUTER -DMBED_CONF_MBED_MESH_API_6LOWPAN_ND_PANID_FILTER=0xffff -DMBED_CONF_MBED_MESH [... cut ...]
    + FullyQualifiedErrorId : NativeCommandFailed
```

This was using PowerShell, on a command prompt, the input is just truncated.

It appears that the command length limit on Windows is being reached and causing compilation to fail.

https://support.microsoft.com/en-us/help/830473/command-prompt-cmd-exe-command-line-string-limitation

By removing some of the `-D` definitions I was able to get the compiler to run but it failed due to undefined symbols (as expected).

### Issue request type

- [ ] Enhancement
- [x] Bug

nat45928, 2020-08-27:
Looks like this will be resolved here: https://github.com/ARMmbed/mbed-os/pull/13427

0xc0170, 2020-09-01:
Thanks @nat45928 for testing. This is correct, we hit the issue with path length. We hit some issues with cmake and armclang that needs fixing in CMake.

For now, just fetch the response files and use it with gcc arm - they work.

0xc0170, 2020-09-04:
I'll close this, as the response files are going to be merged soon (this is not a problem of tools anyway).

@nat45928 feel free to continue testing and reporting issues.

closed 2020-09-04 by 0xc0170

nat45928, 2020-09-08:
Just for completeness, the response files commit in mbed-os did the trick. Compliation works on windows now.