ARMmbed/mbed-tools #67: docs: Building the project

urutva, 2020-08-25:
### Description

The project build instructions https://github.com/ARMmbed/mbed-tools/blob/master/docs/howto/mbed-tools-howto.md#build-the-project doesn't mention checking out `feature-cmake` branch of mbed-os. Without this step build fails with 
```
CMake Error at CMakeLists.txt:10 (add_subdirectory):
  The source directory

    /Users/devran01/Documents/devaraj/working/tmp/tmp/mbed-os-example-blinky/mbed-os

  does not contain a CMakeLists.txt file.

CMake Error at CMakeLists.txt:16 (mbed_configure_app_target):
  Unknown CMake command "mbed_configure_app_target".

-- Configuring incomplete, errors occurred!
```

### Issue request type

- [ ] Enhancement
- [ ] Bug

Patater, 2020-08-25:
This problem will naturally go away when `feature-cmake` is merged, but remains an issue until then.

Patater, 2020-11-13:
`feature-cmake` has been merged. We should be all good now. Re-open if not.

closed 2020-11-13 by Patater