ARMmbed/mbed-tools #96: Changing toolchain for building does not work

hugueskamba, 2020-10-26:
### Description

Steps to reproduce:
1. Build `mbed-os-example-blinky` with `mbedtools build -t GCC_ARM -m K64F`
1. After the build is completed, run `mbedtools build -t ARM -m K64F -c`

Expected:
The application is rebuilt again from scratch with the ARM toolchain

Actual:
The application does not rebuild.
```
Generating Mbed config...
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/hugkam01/projects/mbed/examples/mbed-os-example-blinky/cmake_build
ninja: no work to do.
```

### Issue request type

- [ ] Enhancement
- [x] Bug

rwalton-arm, 2020-10-26:
This will be fixed in the next release. Commits https://github.com/ARMmbed/mbed-tools/commit/aed70c648c48c9c2407785b5c1817b9919e990fa and https://github.com/ARMmbed/mbed-tools/commit/f6af3b99eafb46a60acc6efa5e49d73077223647 fix this.

Patater, 2020-10-28:
Closing as fixed on master branch

closed 2020-10-28 by Patater