ARMmbed/mbed-tools #116: config: failure to override target attributes

rajkan01, 2020-11-16:
### Description
`mbed-tools` fails to parse `mbed_app.json` when `target` attributes are overridden.

Steps to reproduce:
1. Clone https://github.com/ARMmbed/mbed-os-example-wifi/tree/add_cmakelists
1. Ensure the example project uses the master branch of Mbed OS (https://github.com/ARMmbed/mbed-os)
1. Run `mbedtools build -t <TOOLCHAIN> -m DISCO_L475VG_IOT01A -c`

Expect result:
The application is successfully built.

Actual result:
```
Configuring project and generating build system...
Traceback (most recent call last):
  File "/home/arm/.local/bin/mbedtools", line 11, in <module>
    sys.exit(cli())
  File "/home/arm/.local/lib/python3.6/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/arm/.local/lib/python3.6/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/arm/.local/lib/python3.6/site-packages/mbed_tools/cli/main.py", line 37, in invoke
    super().invoke(context)
  File "/home/arm/.local/lib/python3.6/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/arm/.local/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/arm/.local/lib/python3.6/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/arm/.local/lib/python3.6/site-packages/mbed_tools/cli/build.py", line 57, in build
    generate_config(mbed_target.upper(), toolchain, program)
  File "/home/arm/.local/lib/python3.6/site-packages/mbed_tools/build/config.py", line 27, in generate_config
    config = assemble_config(target_build_attributes, program.root, program.files.app_config_file)
  File "/home/arm/.local/lib/python3.6/site-packages/mbed_tools/build/_internal/config/assemble_build_config.py", line 23, in assemble_config
    return _assemble_config_from_sources_and_lib_files(target_attributes, mbed_lib_files, mbed_app_file)
  File "/home/arm/.local/lib/python3.6/site-packages/mbed_tools/build/_internal/config/assemble_build_config.py", line 47, in _assemble_config_from_sources_and_lib_files
    return Config.from_sources(all_sources)
  File "/home/arm/.local/lib/python3.6/site-packages/mbed_tools/build/_internal/config/config.py", line 119, in from_sources
    _update_config_option(config, key, value, source)
  File "/home/arm/.local/lib/python3.6/site-packages/mbed_tools/build/_internal/config/config.py", line 134, in _update_config_option
    f"Can't update option which does not exist."
ValueError: Can't update option which does not exist. Attempting to set **'ism43362.provide-default'** to 'True' in 'File: /home/arm/projects/mbed/examples/mbed-os-example-wifi/mbed_app.json'
```

### Issue request type

- [ ] Enhancement
- [x] Bug

LDong-Arm, 2020-11-16:
Looks like #109

urutva, 2020-12-01:
The PR https://github.com/ARMmbed/mbed-tools/pull/137 fixes the issue https://github.com/ARMmbed/mbed-tools/issues/109 and not https://github.com/ARMmbed/mbed-tools/issues/116. The issue details is mentioned incorrectly in the PR.

urutva, 2020-12-02:
The issue cannot be reproduced with latest mbed-tools (4.0.0). Closing the issue.

closed 2020-12-02 by urutva