ARMmbed/mbed-tools #129: mbed-tools doesn't parse curstom_targets.json

jrobeson, 2020-11-27:
### Description

custom_targets.json is not read

mbed-tools: 3.5.0

### Issue request type

- [ ] Enhancement
- [x] Bug

Patater, 2020-11-30:
Hi @jrobeson 

Thanks for the bug report. I can confirm this is not implemented yet and that we'll want the ability in `mbed-tools` to use custom targets.

boraozgen, 2020-12-10:
This is a dealbreaker for us.

loverdeg, 2020-12-21:
Need this fixed

rwalton-arm, 2021-01-13:
We're working on this now. There was some delay due to the holidays, but we should have this feature merged in the next couple of weeks.

ladislas, 2021-01-13:
> I'll update the issues when the feature is merged. Should be within the next couple of weeks.

Couple, like one or two, or more like 4 or 6? ;) 

if you need help, please let us know! We're eager to help :)

rwalton-arm, 2021-01-13:
> > I'll update the issues when the feature is merged. Should be within the next couple of weeks.
> 
> Couple, like one or two, or more like 4 or 6? ;)
> 
> if you need help, please let us know! We're eager to help :)

I would say ~2 weeks. 4 or 6 weeks would be a "few weeks" and not a "couple" ;). Sorry you've had to wait a while for this feature. As I mentioned earlier, holidays have caused some delay in completing the work. The `mbed-tools` side of this is basically done, as you can see in #160
 
The remaining issues are:
* Finding a user friendly way to add the custom target's source code to mbed's CMake build system without breaking the build for standard targets. 
* Ensuring the correct target directory is pulled into the build by CMake, based on the filtering rules defined in `custom_targets.json`.

After we come up with a solution for these workflow issues we'll get the `mbed-tools` support merged so you can try it out.

ladislas, 2021-01-14:
> Finding a user friendly way to add the custom target's source code to mbed's CMake build system without breaking the build for standard targets.

@rwalton-arm in our project (using USCRPL/mbed-cmake), we have our custom targets at the root of the project with a CMakeLists.txt files in each of those.

The directory structure:

```
targets/TARGET_LEKA_V1_0_DEV
├── CMakeLists.txt
├── PeripheralNames.h
├── PeripheralPins.c
├── PinNames.h
├── PinNames.orig
└── system_clock.c
```

The CMakeLists.txt: https://github.com/ladislas/mbed-cmake-template/blob/develop/targets/TARGET_DISCO_ORIGINAL/CMakeLists.txt#L5-L18

```cmake
add_library(target_DISCO_ORIGINAL STATIC)

target_sources(target_DISCO_ORIGINAL
  PRIVATE
	PeripheralPins.c
	system_clock.c
)

target_include_directories(target_DISCO_ORIGINAL
  PUBLIC
    ${CMAKE_CURRENT_SOURCE_DIR}
)

target_link_libraries(target_DISCO_ORIGINAL mbed-os)
```

We use different build directories to build with our different targets. When the target is a custom target, we call our config scripts with `-x NAME_OF_TARGET`.

Then we have a simple function to parse the name and link the application code with the right target: https://github.com/ladislas/mbed-cmake-template/blob/develop/src/CMakeLists.txt

```cmake
# in main CMakeLists.txt
target_link_custom_boards(my_app_target)
```

https://github.com/ladislas/mbed-cmake-template/blob/develop/cmake/utils/Utils.cmake#L135-L145

```
# and in utils.cmake
function (target_link_custom_boards TARGET)
	# List of custom targets we use
	set(AVAILABLE_TARGET_BOARDS DISCO_ORIGINAL LEKA_V1_0_DEV)
	if (${TARGET_BOARD} MATCHES "\-x")
		string(REPLACE "\-x " "" CUSTOM_BOARD ${TARGET_BOARD})
		foreach(board IN ITEMS ${AVAILABLE_TARGET_BOARDS})
			if (${CUSTOM_BOARD} MATCHES ${board})
				message(STATUS "Adding ${board} library dependency to ${TARGET}")
				target_link_libraries(${TARGET} target_${board})
			endif()
		endforeach()
	endif()
endfunction()
```

If you're not using a custom target, then you don't have the -x, then the function does nothing and it just uses the mbed target.

The full example is available here: https://github.com/ladislas/mbed-cmake-template

What do you mean by?

> Ensuring the correct target directory is pulled into the build by CMake, based on the filtering rules defined in custom_targets.json.

closed 2021-02-05 by Patater