ARMmbed/mbed-tools #132: Use of .mbed file

ladislas, 2020-11-28:
### Description

With the former mbed-cli, one could add a `.mbed` file to the root of the project to setup the following:

```
ROOT=.
MBED_OS_DIR=./extern/mbed-os
TARGET=DISCO_F769NI
TOOLCHAIN=GCC_ARM
```

The `ROOT` and `MBED_OS_DIR` are really useful for project with custom/specific directory organisation.

Ref: https://github.com/ARMmbed/mbed-cli/blob/76c92d07f77c8571e734e4ea122d51e93489893b/mbed/mbed.py#L1899

I'm wondering if it's something that is being worked on or will be worked on in the future? If needed, I could make a PR for this feature.

### Issue request type

- [x] Enhancement
- [ ] Bug

Patater, 2020-11-30:
Hi @ladislas,

We don't currently intend to bring back the `.mbed` file and aren't doing work to support it. It's good to hear about concrete use cases for the file, though.

If we can determine a CMake-native way to specify where to find mbed-os, this would be the preferable path forward. We'd like to leverage CMake wherever possible and keep mbed-tools focused on json->CMake translation, .lib file management, and providing a user-friendly veneer over what can be done with the CMake CLI directly.

Perhaps editing CMakeLists.txt to use `set(MBED_ROOT ${CMAKE_CURRENT_SOURCE_DIR}/extern/mbed-os CACHE INTERNAL "")` instead of `set(MBED_ROOT ${CMAKE_CURRENT_SOURCE_DIR}/mbed-os CACHE INTERNAL "")` would be suitable with a bit more work. Currently, this would not work as the the `mbedtools configure` command looks for an `mbed-os.lib` file. We would need to remove that check and change how the command works so that one can still translate the Mbed json config into a CMake-readable form, no matter where `mbed-os` is located within the application. It's likely we can read the location of `mbed-os` from the CMake cache and use that information as needed to perform the json->CMake translation. Initiating `configure` from CMake itself, rather than the Python CLI, is also something we've been considering, which would simplify communication of the location of `mbed-os` between CMake and the Python code.

ladislas, 2020-11-30:
> If we can determine a CMake-native way to specify where to find mbed-os, this would be the preferable path forward. We'd like to leverage CMake wherever possible and keep mbed-tools focused on [...]

@Patater I'm happy to hear about that! More CMake is always welcome :) 

> Perhaps editing CMakeLists.txt to use `set(MBED_ROOT ${CMAKE_CURRENT_SOURCE_DIR}/extern/mbed-os CACHE INTERNAL "")` instead of `set(MBED_ROOT ${CMAKE_CURRENT_SOURCE_DIR}/mbed-os CACHE INTERNAL "")` would be suitable with a bit more work.

Yes, that's what we are currently doing.

> Initiating `configure` from CMake itself, rather than the Python CLI, is also something we've been considering, which would simplify communication of the location of `mbed-os` between CMake and the Python code.

That would be really interesting and maybe answer @rwalton-arm's question (https://github.com/ARMmbed/mbed-tools/issues/133#issuecomment-735831040) about the root of the project. In this case it's easier, it's where the project `CMakeLists.txt` is.

Patater, 2020-12-03:
For `ROOT=`, it looks like the best route is to use CMake's idea of project bounds. Maybe the Mbed OS application is a part of a larger project, maybe you are making a library, but the "top-llevel root" would be `CMAKE_SOURCE_DIR`. The "root" of the Mbed OS application, within the Mbed OS application's CMake, would be `CMAKE_PROJECT_SOURCE_DIR`. As mentioned, this will require making the `mbedtools configure` command capable of using custom mbed-os locations (e.g. with `--path-to-mbed-os`), instead of always finding it based on the presence of an `mbed-os.lib` file.

For `MBED_OS_DIR=`, the `--path-to-mbed-os` approach currently seems best. This should be useful both now and also later when we make CMake initiate `configure` (as CMake can specify the command line option for the configuration tool).

Patater, 2020-12-03:
@ladislas I've updated the title to reflect what I understand is the underlying ask (of which previously the `.mbed` file was the mechanism to accomplish). If this is not accurate, let me know. If there are other asks I'm missing, we might be best served by raising more issues: one for each ask.

loverdeg, 2020-12-03:
I had a feeling that is what this issue was about.
Yes please add this functionality back in.

loverdeg, 2020-12-03:
Also, we need to be able to point to a local mirror of Mbed-os instead of github. This would be for download speed and availability of github reasons.

ladislas, 2020-12-03:
> I've updated the title to reflect what I understand is the underlying ask (of which previously the .mbed file was the mechanism to accomplish). 

@Patater yes that's right! 

This issue is superseded by #133, so feel free to close if needed.

rwalton-arm, 2020-12-11:
> > I've updated the title to reflect what I understand is the underlying ask (of which previously the .mbed file was the mechanism to accomplish).
> 
> @Patater yes that's right!
> 
> This issue is superseded by #133, so feel free to close if needed.

#133 has been closed, so I'll also close this. Feel free to reopen, or raise another ticket, if there's still an issue with the current behaviour.