ARMmbed/mbed-tools #133: Allow `configure` to not look for mbed-os.lib and provide a path to mbed-os instead

ladislas, 2020-11-29:
### Description

When running `mbed-tools configure`, the absence of `mbed-os.lib` will raise an error.

In our case, mbed-os is located at `$root_of_project/extern/mbed-os`.

It would be nice to be able to tell `mbed-tools` to **not** look for `mbed-os.lib` and instead provide a path the the mbed-os directory.

I'm happy to work on a PR if you'd be interested.
 
### Issue request type

- [x] Enhancement
- [ ] Bug

Patater, 2020-11-30:
@ladislas This sounds like a good feature to add. A PR would be very welcome. Please see our [contributing guide](https://github.com/ARMmbed/mbed-tools/blob/master/CONTRIBUTING.md) for some gotchas on contributing to help you get past the CI gauntlet.

Let me know if I have the design right: when a `--path-to-mbed-os` option is provided to `configure` or `compile`, the configuration action (that translates from json to CMake) would use the given mbed-os path instead of searching for `mbed-os.lib`. When the `--path-to-mbed-os` option is not provided, behavior is as before: an `mbed-os.lib` file is looked for and if it is not present, an error is raised.

rwalton-arm, 2020-11-30:
> @ladislas This sounds like a good feature to add. A PR would be very welcome. Please see our [contributing guide](https://github.com/ARMmbed/mbed-tools/blob/master/CONTRIBUTING.md) for some gotchas [...]

There is at least one other point to consider before implementing this feature. The `mbed-os.lib` is used to determine the root of the project, and we use that knowledge to place `mbed_config.cmake` in a location where mbed's CMake build system expects it to be. So, we would need to consider how we specify the path for `mbed_config.cmake`. Maybe instead of generating it at the repository root, we could always put it in the `mbed-os` directory, and tell mbed's CMake system to look for it there (using `set(MBED_CONFIG_PATH ${MBED_ROOT}/.mbedbuild CACHE INTERNAL "")`), which would make the system more robust when dealing with custom project layouts.

ladislas, 2020-11-30:
> The mbed-os.lib is used to determine the root of the project, and we use that knowledge to place mbed_config.cmake in a location where mbed's CMake build system expects it to be.

@rwalton-arm yes, that was also kind of my point in #132 about the location of root and mbed-os.
The `--path-to-mbed-os` could also require a `--project-root`

@Patater I've created a draft PR #136 (a real draft) with what I had in mind. If this looks like a good way to handle things, I'll go this route. I'm not a python expert, so I'm open to all the suggestions as obvious things for you might not be so obvious for me :) 

I haven't added `compile` yet as we run `cmake --build` directly, but I'll add it as well.

commit 9ad9be8e (2020-12-10): Allow users to provide mbed-os path

This will turn off the check for existing mbed-os.lib file

closes #133
closed 2020-12-10 by Patater