ARMmbed/mbed-tools #143: Program map HTML representation is unreadable in most cases

AGlass0fMilk, 2020-12-03:
### Description

I'm not sure if this is a bug in the new tools but using the old tools (which I am assuming use the same HTML generation code) the "no-change" highlighting color is unreadable. The contrast is so low it's an attack on my eyeballs. White text on light yellow?  

:hocho:
__:eye: __

See screenshot below:

[image]

### Issue request type

- [ ] Enhancement
- [X] Bug

Patater, 2021-02-11:
I'm not sure what this is or where it comes from. How can I reproduce this?

AGlass0fMilk, 2021-02-11:
> I'm not sure what this is or where it comes from. How can I reproduce this?

This should reproduce it:

1. Clone an mbed example, eg: mbed-os-example-blinky
2. Build the example
3. Without changing anything, build the example again
4. Open BUILD/<TARGET>/<TOOLCHAIN>/mbed-os-example-blinky_map.html
5. Put on some sunglasses

Patater, 2021-03-10:
I don't see any html files generated with the new tools.

After doing `mbed-tools compile -t GCC_ARM -m K64F`, `find cmake_build/ -name \*.html` fails to return any results.

Is creating an html memory map something we want a homemade Mbed tool for? Isn't there something else nice out there (hopefully something that doesn't make our eyes bleed)?

Patater, 2021-03-10:
Closing as this isn't an issue with mbed-tools, but if we want a memory map tool that outputs html please raise an enhancement issue.

closed 2021-03-10 by Patater