ARMmbed/mbed-tools #145: Add possibility to use external scripts to detect target, flash firmware, reset board and discover serial port for custom boards

vznncv, 2020-12-05:
### Description

The old `mbed-cli` tool supports [custom boards](https://os.mbed.com/blog/entry/Custom-and-community-board-support/) like [Bluepill](https://os.mbed.com/users/hudakz/code/mbed-os-bluepill/), but it doesn't allow to flash firmware and run [greentea](https://os.mbed.com/docs/mbed-os/v6.5/debug-test/greentea-for-testing-applications.html) tests on them, as some hardware debugger/programmers (like stlink v2) aren't supported or don't have virtual COM port functionality.  
Could you please add a hook for external user scripts/programs that can implement target detection, firmware uploading, board resetting and serial port discovery itself for program uploading and greentea testing.

I have an example of such approach to run greentea tests with `mbed-cli` tool on Bluepill target using stlink v2 and external UART adapter, but it uses monkey patching to add this functionality: https://github.com/vznncv/vznncv-mbed-greentea

### Issue request type

- [x] Enhancement
- [ ] Bug

Patater, 2020-12-08:
This looks like a promising approach, using CMake to hook in flashing and debugging functionality https://github.com/ARMmbed/mbed-os/pull/14010

vznncv, 2020-12-08:
The approach with CMake provides convent way to upload programs with predefined target name `flash-xxx` and start debugging with `debug-xxx`, but:
1. According documentation ("Further Customization" in UploadMethods.md) you need to modify `mbed-os/tools/cmake/upload_methods` if you want to create new upload method or extend existed one.
2. You need to rebuild project if flash/debug method or its option (gdb port, target id, etc.) is changed.
3. It isn't difficult to write simple bash/python wrapper script that can flash target with openocd, pyocd with existing tools. Additionally many IDE already have integration with OpenOCD/PyOCD.
4. It doesn't solve problems with greentea tests, as this approach doesn't have integration with other mbed tools and doesn't provide way to resolve serial port, that can be a separate device.

So I think that it would be better to have some extension points in the mbed-tools itself (especially due item 4), rather than in the build system.

MarceloSalazar, 2021-03-11:
I believe this would be very useful to help us to have similar functionality compared to the [plugins](https://github.com/ARMmbed/mbed-os-tools/tree/master/packages/mbed-host-tests/mbed_host_tests/host_tests_plugins) for the old mbed-os-tools