ARMmbed/mbed-tools #147: memap.py causes dependency issues

boraozgen, 2020-12-08:
### Description

Hi all,

Finally I had the chance to try out the new CLI. I installed mbed-tools according to the [docs](https://os.mbed.com/docs/mbed-os/v6.5/build-tools/install-or-upgrade.html). After realizing that the usage docs are already outdated with the 4.x release 😢, I managed the compile, but at the end I receive a missing dependency error from `memap.py`. I did a fresh installation in the conda package manager, so the previous Mbed CLI related packages were not installed. I believe this is the core problem. Has anyone else noticed this? I guess the dependencies should be updated if we are still dependent on the packages of the old CLI.

- mbed-tools 4.0.0
- Windows 10
- Python 3.9.0

```
[1/1] Linking CXX executable mbed-os-example-blinky
FAILED: mbed-os-example-blinky mbed-os-example-blinky.link_script.ld
cmd.exe [...]
-- built: C:/Users/oezgen.NANTIS/Desktop/mbed-os-example-blinky/cmake_build/mbed-os-example-blinky.bin
-- built: C:/Users/oezgen.NANTIS/Desktop/mbed-os-example-blinky/cmake_build/mbed-os-example-blinky.hex
Traceback (most recent call last):
  File "C:\Users\oezgen.NANTIS\Desktop\mbed-os-example-blinky\mbed-os\tools\memap.py", line 33, in <module>
    from prettytable import PrettyTable, HEADER
ModuleNotFoundError: No module named 'prettytable'
ninja: build stopped: subcommand failed.
```

### Issue request type

- [ ] Enhancement
- [x] Bug

0xc0170, 2020-12-08:
Hi @boraozgen, thanks for the report. It is in the backlog, we will look at this soon.

I already created requirements.txt for Cmake as PR but that is rather a workaround, so we should reconsider this.

0xc0170, 2020-12-08:
@rajkan01 Please review

boraozgen, 2020-12-08:
Which PR do you mean? This is another question I have: Where is the project tracked? I see some tasks in GitHub Projects, and there are some references in the changelog, of which I have no idea where they are referenced...

multiplemonomials, 2020-12-08:
By the way, I ran into this issue on mbed-cmake too.  What I did was I just had cmake detect whether `prettytable` was installed and, if not, disable memory mapping entirely.  You'll notice I added my super-convenient [CheckPythonPackage module](https://github.com/multiplemonomials/mbed-os/blob/cmake-upload-debug/tools/cmake/upload_methods/CheckPythonPackage.cmake) in the upload methods PR which can be used to do this in like 2 lines of code (like [this](https://github.com/multiplemonomials/mbed-os/blob/3bda9720cf6b1b9daf8c94461b6e8ccf7ad8ee79/tools/cmake/upload_methods/UploadMethodPYOCD.cmake#L12)).

boraozgen, 2020-12-08:
Also this might be a good time to review the memory mapping, which has its own issues, see https://github.com/ARMmbed/mbed-os/issues/10922

0xc0170, 2020-12-08:
> Where is the project tracked? I see some tasks in GitHub Projects, and there are some references in the changelog, of which I have no idea where they are referenced...

That is on of the things to be fixed. We are currently having this tracked internally only. The Github contains technical issues only at the moment.

@boraozgen @multiplemonomials How do you find memmap (besides the bugs, that could be fixed.) ? Is it worth making memmap as own python module that could be installed ?

boraozgen, 2020-12-08:
> That is on of the things to be fixed. We are currently having this tracked internally only. The Github contains technical issues only at the moment.

Would be nice, as I now have multiple issues during my evaluation and don't want to report duplicate stuff you already track internally...

> How do you find memmap (besides the bugs, that could be fixed.) ? Is it worth making memmap as own python module that could be installed ?

What does it do other than to report the memory information in a few lines? Is there a documentation? I would like a way of reporting the module sizes with deeper hierarchy (more than the current one level), but I guess 3rd party tools could be used for that...

0xc0170, 2020-12-08:
If issue is not on Github, raise a new one please.

Quick check for the differences between what toolchain provides and what memmap provides:

GCC ARM "-Wl,--print-memory-usage" flag:

```
Memory region         Used Size  Region Size  %age Used
    m_interrupts:          1 KB         1 KB    100.00%
  m_flash_config:          16 B         16 B    100.00%
          m_text:       42688 B    1047536 B      4.08%
          m_data:         64 KB        64 KB    100.00%
        m_data_2:        191 KB       192 KB     99.48%
```

memmap module:
```

| Module           |         .text |       .data |        .bss |
|------------------|---------------|-------------|-------------|
| [fill]           |     116(+116) |       4(+4) | 2185(+2185) |
| [lib]\c.a        | 12027(+12027) | 2472(+2472) |     89(+89) |
| [lib]\gcc.a      |     760(+760) |       0(+0) |       0(+0) |
| [lib]\misc       |     188(+188) |       4(+4) |     28(+28) |
| [lib]\nosys.a    |       32(+32) |       0(+0) |       0(+0) |
| main.cpp.obj     |       44(+44) |       0(+0) |       0(+0) |
| mbed-os\cmsis    |   8017(+8017) |   168(+168) | 5952(+5952) |
| mbed-os\drivers  |     206(+206) |       0(+0) |       0(+0) |
| mbed-os\hal      |   1697(+1697) |       8(+8) |   115(+115) |
| mbed-os\platform |   7279(+7279) |   260(+260) |   365(+365) |
| mbed-os\rtos     |     214(+214) |       0(+0) |       0(+0) |
| mbed-os\targets  |   9286(+9286) |     36(+36) |   386(+386) |
| Subtotals        | 39866(+39866) | 2952(+2952) | 9120(+9120) |
Total Static RAM memory (data + bss): 12072(+12072) bytes
Total Flash memory (text + data): 42818(+42818) bytes
```

ARMClang "--info=summarysizes,sizes,totals,unused,veneers"

```
Program Size: Code=35512 RO-data=4756 RW-data=292 ZI-data=205684
```

memmap:
```
| Module              |         .text |     .data |          .bss |
|---------------------|---------------|-----------|---------------|
| [lib]\c_w.l         |   4368(+4368) |   16(+16) |     348(+348) |
| [lib]\fz_wm.l       |       26(+26) |     0(+0) |         0(+0) |
| [lib]\libcppabi_w.l |       44(+44) |     0(+0) |         0(+0) |
| anon$$obj.o         |       32(+32) |     0(+0) |   2304(+2304) |
| main.o              |       48(+48) |     0(+0) |         0(+0) |
| mbed-os\cmsis       |   9468(+9468) | 168(+168) |   6609(+6609) |
| mbed-os\drivers     |     534(+534) |     0(+0) |         0(+0) |
| mbed-os\hal         |   1672(+1672) |     8(+8) |     115(+115) |
| mbed-os\platform    |   8122(+8122) |   64(+64) |     357(+357) |
| mbed-os\rtos        |     425(+425) |     0(+0) |         0(+0) |
| mbed-os\targets     | 15389(+15389) |   36(+36) |     341(+341) |
| Subtotals           | 40128(+40128) | 292(+292) | 10074(+10074) |
Total Static RAM memory (data + bss): 10366(+10366) bytes
Total Flash memory (text + data): 40420(+40420) bytes
```

0xc0170, 2020-12-08:
I believe we will use scripts with CMake so requirements will be required anyway. One way to address this would be to add requirements to CMake Mbed OS: https://github.com/ARMmbed/mbed-os/pull/13944.  If we upstream memmap to be own package, it would be just updated. If we add additional script to post process output file, all would be added to requirements. It is now left in between nowhere.

ladislas, 2020-12-08:
> Is it worth making memmap as own python module that could be installed ?

@0xc0170 you're talking about `memap.py` right? not the numpy.me**mm**ap module.

I think it would be nice to have it as an external module or even better, integrate the new version to mbed-tools. I've seen some issues when mbed-os is not at the root of the project, you don't get the nice division between modules.

Being able to work on it with PRs here would be nice.

jrobeson, 2020-12-12:
@ladislas It's not just if mbed-os is at the root, but also and/if mbed_app.json is not at the root.

ladislas, 2020-12-13:
@jrobeson what do you mean?

in my case, mbed_app.json is a the root of the project. mbed-os is located at `./extern/mbed-os`.

memap.py shows only one category for mbed-os: `extern/mbed-os`

jrobeson, 2020-12-13:
If you put `mbed_app.json` somewhere else and then use `--app-config=/path/to/mbed_app.json` it also ruins the memmap display

Patater, 2021-02-11:
Fixed via https://github.com/ARMmbed/mbed-tools/pull/170

closed 2021-02-11 by Patater