ARMmbed/mbed-tools #150: ELF file is generated without extension

boraozgen, 2020-12-09:
### Description

Self-explanatory.

Additionally, when not using Ninja, no elf files are generated. Not an issue if using Ninja is a constraint.

mbed-tools 4.0.0
Windows 10

### Issue request type

- [ ] Enhancement
- [x] Bug

0xc0170, 2020-12-09:
what toolchain do you use (to reproduce this) ? IPlease share exact commands (also what generators you are using).

boraozgen, 2020-12-09:
Toolchain: `arm-none-eabi-gcc.exe (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)`

`mbed-tools configure -m K64F -t GCC_ARM`
`mbed-tools compile -m K64F -t GCC_ARM` -> I believe this uses Ninja.

Resulting build directory:
```
Mode                LastWriteTime         Length Name
----                -------------         ------ ----
d-----       09.12.2020     17:51                CMakeFiles
d-----       09.12.2020     17:50                mbed-os
-a----       09.12.2020     17:51         205868 .ninja_deps
-a----       09.12.2020     17:51          64846 .ninja_log
-a----       09.12.2020     17:50        4150434 build.ninja
-a----       09.12.2020     17:50          18326 CMakeCache.txt
-a----       09.12.2020     15:17           1903 cmake_install.cmake
-a----       09.12.2020     17:50          15621 compile_time_defs.txt
-a----       09.12.2020     17:51        2431720 mbed-os-example-blinky -> extension missing!
-a----       09.12.2020     17:51          43824 mbed-os-example-blinky.bin
-a----       09.12.2020     17:51         123193 mbed-os-example-blinky.hex
-a----       09.12.2020     17:51           4840 mbed-os-example-blinky.link_script.ld
-a----       09.12.2020     17:51        4913361 mbed-os-example-blinky.map
```

0xc0170, 2020-12-10:
I've tested it with ARMClang,  `mbed-os-example-blinky.elf` is created. This means there is a difference between ARMClang and Gcc Arm. 

From ARMClang: https://github.com/Kitware/CMake/blob/master/Modules/Compiler/ARMClang.cmake#L12
I can't find this for Gcc Arm, `CMAKE_EXECUTABLE_SUFFIX` is not set there. We can patch this now locally and find out why in the upstream is not set (Iar has it as well set to .elf).

0xc0170, 2020-12-10:
would you be able to check if Gcc Arm is not really setting `CMAKE_EXECUTABLE_SUFFIX ` and sending a pull request adding suffix for Gcc Arm, to our Gcc Arm  toolchain file?

boraozgen, 2020-12-10:
I guess it makes sense that it is not set in GNU C compiler since in Linux the executables are without extensions... I believe we should set it in our side. I'm gonna look into it, I'm pretty inexperienced with CMake and it might take some time.

boraozgen, 2020-12-10:
I tried setting it in the toolchain file but it did not work.

Then I found this: https://gitlab.kitware.com/cmake/cmake/-/issues/16538

Does this mean we need an additional platform file?

0xc0170, 2020-12-10:
I havent used platform file, it looks like it's not trivial. If there is no other way, then yes unfortunately. but only for Gcc Arm for now.

Patater, 2021-02-11:
This was fixed in Mbed OS's CMake.

closed 2021-02-11 by Patater