ARMmbed/mbed-tools #151: Add test command

boraozgen, 2020-12-09:
### Description

A test command like in CLI 1 should be added with options to run unit tests and Greentea tests separately, possibly with groups and single modules etc. This time it should support custom tests, too.

### Issue request type

- [x] Enhancement
- [ ] Bug

0xc0170, 2020-12-09:
We are working currently on CMake boilerplate in Mbed OS to support this. The tools part will come as well.

Good point about custom tests, it should be possible to do.

ladislas, 2020-12-09:
@0xc0170 looking forward to that as well!

Patater, 2021-04-08:
Just a quick update. With [this mbed-os PR](https://github.com/ARMmbed/mbed-os/pull/14426) we'll be able to build and run mbed-os unit tests with vanilla CTest.

Next would be to get the Greentea tests buildable and runnable from CTest, too.

LDong-Arm, 2021-08-11:
This is now replaced by #304 which contains a breakdown of sub-tasks