ARMmbed/mbed-tools #153: Add possibility to configure toolchain location

boraozgen, 2020-12-09:
### Description

This was possible in CLI 1.

### Issue request type

- [x] Enhancement
- [ ] Bug

rwalton-arm, 2020-12-11:
Hi,

Thanks for trying out the tools and for your feedback. This isn't something we're planning on adding to the Python tools as the toolchain location is managed by CMake. If you want to configure the location you may be able to set environment variables as described [in this stackoverflow post](https://stackoverflow.com/questions/17275348/how-to-specify-new-gcc-path-for-cmake). A possible alternative would be to edit the toolchain file located in `mbed-os/tools/cmake/toolchains/`.

boraozgen, 2020-12-11:
Thanks for clarifying this. Maybe this should be added to the docs to avoid confusion.

Still, I noticed at least one line where the toolchain command is hardcoded:

https://github.com/ARMmbed/mbed-os/blob/36df9d10436487625810a43f922b7fb975e5eec7/CMakeLists.txt#L135

Trying to compile with CLion (where the toolchain location is customizable as you mentioned) still fails at this command.

rwalton-arm, 2021-01-13:
> Thanks for clarifying this. Maybe this should be added to the docs to avoid confusion.
>
> Still, I noticed at least one line where the toolchain command is hardcoded: [...]

That hardcoded toolchain is a bug in mbed-os' CMake. We will look into removing that.

0xc0170, 2021-01-15:
I realized we do not have `arm-none-eabi-cpp` set anywhere, this is a bug and should be fixed. @boraozgen would you be able to send pull request fixing this?

boraozgen, 2021-01-18:
I unfortunately don't have enough knowledge about the tooling to fix this.

Patater, 2021-02-11:
Fixed in Mbed OS in https://github.com/ARMmbed/mbed-os/pull/14199

closed 2021-02-11 by Patater