ARMmbed/mbed-tools #156: Some target settings defined in targets.json cannot be modified from config

rwalton-arm, 2020-12-11:
### Description
Only a limited set of config settings defined in targets.json can be modified in the config system when using `mbed-tools`.

The following settings can be modified:
- `c_lib`
- `printf_lib`
- `extra_labels`
- `features`
- `components`
- `macros`

It probably doesn't make sense for users to modify all of the target settings. Many of them don't end up in the config output at all.  I've tested at least `device_has` and `supported_form_factors` can be modified with the old tools (not sure if these make sense to modify or not, I'd guess form factor doesn't, but I have seen some apps modify the `device_has` field so we will need to support that at least).

### Issue request type

- [ ] Enhancement
- [x] Bug

jrobeson, 2020-12-12:
does mbed_app_start and friends work yet?

rwalton-arm, 2020-12-14:
> does mbed_app_start and friends work yet?

Sorry, the "bootloader overrides" don't work yet. We do have another backlog item to add those in. We will get to that over the next few weeks.

ladislas, 2020-12-14:
@rwalton-arm looking forward to the bootloader settings. It is really needed for our product. We are a bit lost in translation as we are moving to the new CMake build system but need a way to manage the bootloader and application (we are using Pelion)

boraozgen, 2021-05-03:
This is blocking adoption for us. How much effort would this require to be implemented?

0xc0170, 2021-05-03:
@boraozgen https://github.com/ARMmbed/mbed-tools/pull/270 - this does not fix the issue? If not, what we are missing? Can you describe what is blocking your project?

boraozgen, 2021-05-04:
We are using a bootloader with an application. Therefore it is required to shift the application in ROM. [Here](https://os.mbed.com/docs/mbed-os/v6.9/program-setup/creating-and-using-a-bootloader.html) it is explained that `target.mbed_app_start` and `target.mbed_app_size` can be used to do this, and indeed that is what I use with CLI 1. These options are not yet supported in CLI 2.

I see that #270 adds support for `mbed_rom_start` and `mbed_rom_size`, which seem to be doing the same thing, but I'm not sure. I'll try them out.

boraozgen, 2021-05-19:
I tried it today too, like @jrobeson said it does not parse the inputs from `mbed_app.json`.

```
{
    "target_overrides": {
        "*": {
            "target.mbed_rom_start": "0x8020200",
            "target.mbed_rom_size": "0xBE000",
        }
    }
}
```

This does not change anything in `mbed_config.cmake`. Furthermore I noticed that invalid configurations are not detected in `mbed_app.json`, but that's another issue.

ladislas, 2021-05-19:
hitting the same issue working on mcuboot.

jrobeson, 2021-05-20:
I thought i'd post the actual error from mbed-tools now that i found it again:

```
WARNING: You are attempting to override an undefined config parameter `target.mbed_app_start`.
It is an error to override an undefined configuration parameter. Please check your target_overrides are correct.
The parameter `target.mbed_app_start` will not be added to the configuration.
WARNING: You are attempting to override an undefined config parameter `target.mbed_app_size`.
It is an error to override an undefined configuration parameter. Please check your target_overrides are correct.
The parameter `target.mbed_app_size` will not be added to the configuration.
```

jrobeson, 2021-05-20:
ah, i see the problem then.  For me it's the nrf52832 and nrf52840 chips and various boards based on them.

boraozgen, 2021-05-20:
STM32F4 based custom targets and dev boards (e.g. NUCLEO_F429ZI).

I just modified targets.json to define `mbed_rom_start` and `mbed_rom_size` for my target, and the compiler flags appeared in `mbed_config.cmake`. However I cannot override them in `mbed_app.json`. What is the syntax for it?

ladislas, 2021-05-20:
> we decided the best approach was to move everything to targets.json 

@rwalton-arm do you want to handle that PR by PR as they come or does it need to be 1 big PR to rule them all?

rwalton-arm, 2021-05-20:
> However I cannot override them in `mbed_app.json`. What is the syntax for it?

Your syntax above looked correct. I've just tried it myself and I can reproduce the issue. Looks like a bug in the memory region collection.

rwalton-arm, 2021-05-20:
> > we decided the best approach was to move everything to targets.json
> 
> @rwalton-arm do you want to handle that PR by PR as they come or does it need to be 1 big PR to rule them all?

PR by PR would be easier to review, but I guess one PR to rule them all would also work.

ladislas, 2021-05-20:
We've just tried this, and it seems to be working in our case:

```json
{
    "config": {
        "mbed_app_start": {
            "help": "Use a custom application start address",
            "value": "0x08020000",
            "macro_name": "MBED_APP_START"
        }
    },
    "target_overrides": {
        "*": {
            "platform.stdio-baud-rate": 115200,
        }
    }
}
```

boraozgen, 2021-05-20:
@ladislas this generates the flags in mbed_config.cmake but does not seem to shift the image. A quick search for `MBED_CONFIG_DEFINITIONS` shows that these flags are not given to the linker. I'm on 6.9.0, don't know if this changed in 6.10.

ladislas, 2021-05-20:
we are on 6.9 as well. have you made a deep clean before building?
are you using a custom target? (we are)

boraozgen, 2021-05-20:
Deep clean did it, don't know why though. Not a good sign IMHO :/

Nice workaround, thanks!

ladislas, 2021-05-20:
Great! My team mate @YannLocatelli is the real here here 🙏

kalle16lab, 2021-05-21:
Hi! The workaround mentioned by @ladislas woks but I am also currently interested in switching to CLI2 from Mbed Studio and this is also blocking progress on a commercial project.

Any ETA on a pull request to master to fix this?

jrobeson, 2021-05-21:
@kalle16lab  If it's a commercial project, then seems like you should be able assist right?

ladislas, 2021-05-21:
@kalle16lab we are not using mbed studio but mbed cli2

aabc123, 2023-09-08:
Just looking to put a boot loader on my Nucleo F411 which is now running mbed and hitting this issue.  The work-around for an unmanaged boot loader posted above works for offsetting the app address.

Is there a way to make use of the built-in header with this work-around or do we need to manually place the header also?