ARMmbed/mbed-tools #161: Missing package dependencies in pip package

bjanders, 2020-12-22:
### Description

* I have installed python 3.9.1 on macOS 11.1 using brew.
* I installed mbed-tools 4.3.0 using `python3 -m pip install mbed-tools`. (I have since upgraded to 4.4.0)
* I than ran mbed-tools compile
* I got successive errors that the following Python modules where missing:
**  pretty table
** future
** intelhex

After installing the above three packages via pip, everything worked. It seems there are missing package dependencies in the mbed-tools package.

### Issue request type

- [ ] Enhancement
- [x ] Bug

bjanders, 2020-12-22:
It could be that the missing dependencies are actually in some other package(s),  as I cannot find a reference to these in mbed-tools, except that in `travis.yml` the following is run:

```
        - pip install -U 'prettytable<1.0'
        - pip install -U 'future==0.16.0'
        - pip install -U 'Jinja2>=2.10.1,<2.11'
        - pip install -U 'intelhex>=1.3,<=2.2.1'
```

I didn't record the backtrace, so I can't verify right now. I might uninstall the problematic packages to find out.

bjanders, 2020-12-22:
After uninstalling prettytable, future and intelhex, and running `cmake --build cmake_build`:

```
Traceback (most recent call last):
  File ".../test-project/mbed-os/tools/memap.py", line 33, in <module>
    from prettytable import PrettyTable, HEADER
ModuleNotFoundError: No module named 'prettytable'
ninja: build stopped: subcommand failed.

Traceback (most recent call last):
  File "...test-project/mbed-os/tools/memap.py", line 36, in <module>
    from future.utils import with_metaclass
ModuleNotFoundError: No module named 'future'
ninja: build stopped: subcommand failed.

Traceback (most recent call last):
  File ".../test-project/mbed-os/tools/memap.py", line 43, in <module>
    from tools.utils import (
  File ".../test-project/mbed-os/tools/utils.py", line 33, in <module>
    from intelhex import IntelHex
ModuleNotFoundError: No module named 'intelhex'
ninja: build stopped: subcommand failed.
```

rwalton-arm, 2020-12-24:
Hello and thanks for the bug report. 

This occurs because mbed-os CMake executes `memap.py` (a script located in the mbed-os repo) to print the memory map after building. This is a known issue internally. As you already noticed, we work around this in our travis CI by manually installing the dependencies with `pip`. We can track the issue here for visibility, but it's not something we will fix in the `mbed-tools` package directly. We don't want to list non-package dependencies in `mbed-tools`' `setup.py`. We will probably handle this by creating an "Mbed installer", which will ensure all dependencies and tools are installed up front so there are no nasty surprises at build time.  At the moment the workaround is to manually install the requirements from https://github.com/ARMmbed/mbed-os/blob/master/tools/cmake/requirements.txt

One of the issues with our old tools was that the `mbed-cli` sometimes tried to install python dependencies from mbed-os when the user executed a command. It generally did this in an unsafe way, that is when it didn't just crash because the dependencies were missing. This was/is a terrible user experience. We do need to ensure all dependencies required by our tooling and CMake scripts are installed before users start running commands.

bjanders, 2020-12-24:
Thanks for the explanation. If/when it is mentioned in the documentation, then it is good enough for me. Feel free to close this for my part whenever you see fit.

Patater, 2021-01-26:
Hi @bjanders 

We've just added the dependencies from mbed-os to mbed-tools itself to help avoid this usability problem you ran into, via #170. This is a bit of a hack though, so we've filed #171 for covering how we want to fix it properly in the future.

Hope this helps!

closed 2021-01-26 by Patater