ARMmbed/mbed-tools #163: Configure STM32WB55RG for Mbed CLI2

kvkc97, 2021-01-01:
Hi,
I am using Mbed CLI2 to add a custom target STM32WB55RGV6. My system is Windows10 and I am using latest Mbed version. Since Mbed CLI2 supports boards K64F, DISCO_L475VG_IOT01A and NRF52840_DK only, I tried to add a custom target for my requirement following the given link. 
https://os.mbed.com/docs/mbed-os/v6.6/program-setup/adding-and-configuring-targets.html
When I tried to configure the new target, error is obtained that 'Attributes are not found'.  Is this the right step to be followed or is there some other steps to be followed?
Also since the version of Mbed CLI2 is beta, is there any other option to add a custom board?

Thanks & Regards
Keerthi

ladislas, 2021-01-04:
@kvkc97 custom targets are not yet supported (see #129) but a PR is on its way to add support (see #160)

kvkc97, 2021-01-04:
I have tried the above mentioned method but it generates the same error.

0xc0170, 2021-01-05:
> I have tried the above mentioned method but it generates the same error.

Can you describe what have you tried (also did you fetch pr #160 ?) ? The detailed report will help PR 160 to proceed.

kvkc97, 2021-01-05:
I followed pr #160  and added add_subdirectory(<CUSTOMTARGETDIR>) in the application level CMakeLists.txt. This generates the same error that is "Target attributes are not found"

rwalton-arm, 2021-01-08:
Hi @kvkc97 could you post your custom_targets.json? It's a possibility that the docs are incorrect, or I've messed something up as I've only tried it with one example so far.

kvkc97, 2021-01-08:
[image]
[image]
I have added the custom_targets.json file.

rwalton-arm, 2021-01-08:
It looks like you haven't named your target from the screenshot. The key in the outer dictionary should be the target name, e.g
```
{
"MYTARGET": {
        "inherits": ["Target"]
...
```

You then need to pass the name to  the `build` command e.g 
```
mbedtools build -m MYTARGET -t GCC_ARM
```

kvkc97, 2021-01-08:
I have added the target name in custom_targets.json file. Since build command is not available in mbed-tools, i used configure command as follows.
mbed-tools configure -m MYTARGET -t GCC_ARM
It generated the same error, that is Target attributes are missing

rwalton-arm, 2021-01-11:
> I have added the target name in custom_targets.json file. Since build command is not available in mbed-tools, i used configure command as follows.
> mbed-tools configure -m MYTARGET -t GCC_ARM
> It generated the same error, that is "Target attributes for MYTARGET not found."

Which version of mbed-tools are you using? We've had a build command available for a while now.

kvkc97, 2021-01-11:
I have used both compile and configure commands but both generate error. I have attached the screenshot of the same.
[image]

rwalton-arm, 2021-01-11:
> I have used both compile and configure commands but both generate error. I have attached the screenshot of the same.
> [image]

OK. I'll take a look. Could you post a copy/paste of your custom_targets.json so I can try it here? Also do you have `custom_targets.json` in the root of your project when you run the command?

kvkc97, 2021-01-11:
I have attached the  custom_targets.json file below. And the file is saved in the root folder along with main.cpp and other library files.

{
"MYTARGET": {
        "inherits": ["Target"],
		"core": "Cortex-M4F",
		"public": false,
		"macros": ["USE_HAL_DRIVER", "USE_FULL_LL_DRIVER", "TRANSACTION_QUEUE_SIZE_SPI=2", "MBED_TICKLESS", "EXTRA_IDLE_STACK_REQUIRED", "STM32WB55xx", 
			"MBEDTLS_CONFIG_HW_SUPPORT"],
		"extra_labels": ["STM", "STM32WB", "CORDIO", "STM32WB55xx"],
		"features": ["BLE"],
		"config": {
            "lse_available": {
                "help": "Define if a Low Speed External xtal (LSE) is available on the board (0 = No, 1 = Yes). If Yes, the LSE will be used to clock the RTC, LPUART, ... otherwise the Low Speed Internal clock (LSI) will be used",
                "value": "1"
            },
            "lpuart_clock_source": {
                "help": "Define the LPUART clock source. Mask values: USE_LPUART_CLK_LSE, USE_LPUART_CLK_PCLK1, USE_LPUART_CLK_HSI",
                "value": "USE_LPUART_CLK_LSE|USE_LPUART_CLK_PCLK1"
            },
            "stdio_uart_tx": {
                "help": "default TX STDIO pins is defined in PinNames.h file, but it can be overridden"
            },
            "stdio_uart_rx": {
                "help": "default RX STDIO pins is defined in PinNames.h file, but it can be overridden"
            },
            "lpticker_delay_ticks": {
                "help": "https://os.mbed.com/docs/latest/porting/low-power-ticker.html",
                "value": 1,
                "macro_name": "LPTICKER_DELAY_TICKS"
            },
            "lpticker_lptim_clock": {
                "help": "Default value for LPTIM clock (lpticker_lptim == 1). Value is the dividing factor. Choose 1, 2 or 4",
                "value": 1
            },
			"lpticker_lptim": {
                "help": "This target supports LPTIM. Set value 1 to use LPTIM for LPTICKER, or 0 to use RTC wakeup timer",
                "value": 1
            },
            "lse_drive_load_level": {
                "help": "HSE drive load level  RCC_LSEDRIVE_LOW | RCC_LSEDRIVE_MEDIUMLOW | RCC_LSEDRIVE_MEDIUMHIGH | RCC_LSEDRIVE_HIGH",
                "value": "RCC_LSEDRIVE_LOW"
            }
        },
		"overrides": {
            "deep-sleep-latency": 4,
            "init-us-ticker-at-boot": true,
			"lpticker_delay_ticks": 0
        },
		"device_has": ["USTICKER", "LPTICKER", "RTC", "ANALOGIN", "I2C", "I2CSLAVE", "I2C_ASYNCH", "INTERRUPTIN", "PORTIN", "PORTINOUT", "PORTOUT", "PWMOUT",
            "SERIAL", "SERIAL_FC", "SLEEP", "SPI", "SPISLAVE", "SPI_ASYNCH", "STDIO_MESSAGES", "WATCHDOG", "RESET_REASON", "CRC", "FLASH", "MPU", 
			"SERIAL_ASYNCH", "TRNG", "USBDEVICE"],
		"supported_toolchains": ["ARM", "IAR", "GCC_ARM"],
		"device_name": "STM32WB55RGVx",
		"OUTPUT_EXT": "bin",
		"supported_c_libs": {
            "arm": ["std", "small"],
            "gcc_arm": ["std", "small"],
            "iar": ["std"]
        },
		"bootloader_supported": true,
		"supported_form_factors": ["ARDUINO", "MORPHO"],
		"supported_application_profiles": ["full", "bare-metal"],
	}
}

rwalton-arm, 2021-01-11:
Thanks. Your target is marked `public: false` so the `targets.json` parser ignores it. If you set `public: true` it should find the target attributes. I'm still getting another error from CMake to do with include directories, but that could be just my test env as I didn't add any source files for the custom target. Could you give it another try with `public: true` and let me know what happens? Thanks.

kvkc97, 2021-01-11:
Thanks. I edited custom target file to public:true but same error is generated.

rwalton-arm, 2021-01-11:
Interesting. The error I'm seeing now is:
```
mbedtools --traceback compile -t GCC_ARM -m MYTARGET -c
Configuring project and generating build system...
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at mbed-os/connectivity/drivers/mbedtls/TARGET_STM/TARGET_STM32WB55xx/CMakeLists.txt:4 (target_include_directories):
  Cannot specify include directories for target "mbed-mbedtls-stm" which is
  not built by this project.

...
mbed_tools.build.exceptions.MbedBuildError: CMake invocation failed!
```

Oh, I also had to remove the trailing comma on this line: `"supported_application_profiles": ["full", "bare-metal"],` as it complained the file was invalid JSON with the trailing comma. But I'd expect you to see a different error if that was the case. Can you run `mbedtools --traceback compile ...` and post the output here? Thanks.

kvkc97, 2021-01-11:
The following is the output I got while giving the mentioned command.
[image]
I have added the custom target json file outside the mbed os libraray and in the root folder of the program. That is correct right?

rwalton-arm, 2021-01-11:
> The following is the output I got while giving the mentioned command.
> [image]
> I have added the custom target json file outside the mbed os libraray and in the root folder of the program. That is correct right?

Yeah, it sounds like you have the `custom_targets.json` in the correct place. 

You need to run your full build command with the `-c` flag, as you already have a build folder cached:
```
mbedtools --traceback compile -t GCC_ARM -m MYTARGET -c
```
Sorry that wasn't clear from my last post.

Other useful information would be the output of `mbedtools --version` As I suspect you may not be running the right version of `mbed-tools`. Did you create a new python virtualenv and do a local `pip install` from my branch https://github.com/rwalton-arm/mbed-tools/tree/dev/rwalton-arm/custom-targets-json? You'll also need to ensure the virtualenv is activated when you run the commands so the correct version of `mbed-tools` is prepended to your `PATH`.

kvkc97, 2021-01-11:
Sorry about previous post, I got confused a bit. I have attached the output of the above command and mbedtools version below. Version is 4.5.0.
[image]
I have not used virtual env currently in my system for mbed cli2. But after installing mbed cli2, I had installed mbed cli1 using virtual env but then removed cli1. Should I reinstall mbed cli2 in a virtual env?

rwalton-arm, 2021-01-12:
> Should I reinstall mbed cli2 in a virtual env?

You do appear to be using PyPI release 4.5.0 at the moment. If you want to try out custom targets you will need to do a local `pip install` of `mbed-tools` from my development branch. 

It's worth noting that the custom targets feature is not guaranteed to work in a sensible way yet. It might be easier for you to wait until we merge #160 and create an official release. I won't be able to provide any support on managing your Python installations or virtual environments if something goes wrong during the installation process.

If you still want to try my `custom_targets.json` branch you will need to:

* Clone my fork of the `mbed-tools` repo: `git clone https://github.com/rwalton-arm/mbed-tools.git`
* Check out the `custom-targets-json` branch: `git checkout dev/rwalton-arm/custom-targets-json`
* Create a new virtual environment and activate it
* Ensuring the virtual env is activated, run `python -m pip install .` from the root of the repo (where `setup.py` is located)

That should install the dev version of `mbed-tools` containing the `custom_targets.json` code. If you want to go back to the release version at any point, you can just deactivate the virtual environment.

See [the Python docs](https://docs.python.org/3/tutorial/venv.html) for more information on setting up virtual environments and how they work.

kvkc97, 2021-01-12:
By when can we expect an official release?

rwalton-arm, 2021-01-13:
> By when can we expect an official release?

I'll update the issues when the feature is merged. Should be within the next couple of weeks.

ladislas, 2021-01-13:
> I'll update the issues when the feature is merged. Should be within the next couple of weeks.

Couple, like one or two, or more like 4 or 6? ;)

Patater, 2021-02-10:
Hi @kvkc97

Sorry for not updating this issue immediately. We now support custom targets as of v5.5.0 (2021-02-06), but any newer version should work, too. Let us know how you get on with this.

Thanks!

kvkc97, 2021-03-01:
Hi,

Could you provide any documentations for the same

Patater, 2021-03-01:
@kvkc97 Please refer to https://github.com/ARMmbed/mbed-os-5-docs/pull/1407 where we are getting the documentation put together. It should be merged and published to our website within a few weeks.

kvkc97, 2021-03-03:
Hi,

Custom target is successfully added. Thanks for the support.

Regards
Keerthi

kvkc97, 2021-05-03:
I would like add some more queries to this issue. I have added the custom target successfully and now want to add hal files to it. Currently, my TARGET_IMAGINORYBOARD file contains only the peripheral pin files. Is there a specific format to add the hal files to it?

0xc0170, 2021-05-04:
If you only use mbed-cli2, add files to CMakeLists.txt (mbed-cli1 requires specific directory layout, not mbed-cli2).

kvkc97, 2021-05-04:
I use only mbed-cli2 and attached the hal files. I have a linker along with my files and tried to include it in my CMakeLists file by the following statement.
mbed_set_linker_script(mbed-stm32wb55RG ${CMAKE_CURRENT_SOURCE_DIR}/${LINKER_FILE})

On compiling, an error is generated stating that mbed_set_linker_script is not identified. On commenting the above statement in the CMakeLists file, linking error is generated. 

How can I add the linker file to my CMakeLists file? I have attached the file below.
[image]

0xc0170, 2021-05-17:
> mbed_set_linker_script(mbed-stm32wb55RG ${CMAKE_CURRENT_SOURCE_DIR}/${LINKER_FILE})

Include the cmake file where the definition is (should be in `tools/cmake/mbed_set_linker_script.cmake`)