ARMmbed/mbed-tools #178: Missing --depth option for deploy / new commands

Azq2, 2021-02-06:
### Description

Old tools `mbed-cli` has amazing option --depth for more fast and efficient downloading mbed-os sources, but new tools `mbed-tools` don't have this option.

But this option very useful on limited and slow internet connections. And takes up less space on hdd.

### Issue request type

- [x] Enhancement
- [ ] Bug

Azq2, 2021-02-06:
71% faster
84% less internet traffic

```
$ time git clone https://github.com/ARMmbed/mbed-os full
Cloning into 'full'...
remote: Enumerating objects: 2, done.
remote: Counting objects: 100% (2/2), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 392418 (delta 0), reused 0 (delta 0), pack-reused 392416
Receiving objects: 100% (392418/392418), 413.66 MiB | 9.08 MiB/s, done.
Resolving deltas: 100% (274014/274014), done.
Checking out files: 100% (16844/16844), done.

real	1m49,666s
user	0m56,212s
sys	0m7,043s
```

```
$ time git clone https://github.com/ARMmbed/mbed-os --depth 1 depth1
Cloning into 'depth1'...
remote: Enumerating objects: 18765, done.
remote: Counting objects: 100% (18765/18765), done.
remote: Compressing objects: 100% (14501/14501), done.
remote: Total 18765 (delta 8175), reused 8053 (delta 3476), pack-reused 0
Receiving objects: 100% (18765/18765), 62.77 MiB | 7.67 MiB/s, done.
Resolving deltas: 100% (8175/8175), done.
Checking out files: 100% (16844/16844), done.

real	0m31,639s
user	0m8,406s
sys	0m2,625s
```

31% less space on hdd
```
1,1G	./full
758M	./depth1
```

ladislas, 2021-02-08:
+1 for that! 

I also find that `curl` can be even faster sometimes:

```
export VERSION=master 
curl -O -L https://github.com/ARMmbed/mbed-os/archive/$(VERSION).tar.gz
```

Patater, 2021-02-11:
For `deploy`, lets try always going with `--depth 1`, instead of adding an option.

commit de0dcbc2 (2021-02-24): project: Shallow clone repos

Without shallow cloning, cloning Mbed OS can take quite a long while.
Clone with a depth of 1 when resolving library dependencies in order to
speed up cloning. A downside of this approach is that the resulting
repository will not contain other git references than the initial one,
but this can be worked around by performing a subsequent fetch using
`git fetch`.

Example performance improvement on one person's internet connection:
- Old speed for Mbed OS cloning: 1 minute 12.19 seconds
- New speed for Mbed OS cloning: 26.57 seconds

Fixes #178
commit 9b26770e (2021-03-02): project: Shallow clone repos

Without shallow cloning, cloning Mbed OS can take quite a long while.
Clone with a depth of 1 when resolving library dependencies in order to
speed up cloning. A downside of this approach is that the resulting
repository will not contain other git references than the initial one,
but this can be worked around by performing a subsequent fetch using
`git fetch`.

Example performance improvement on one person's internet connection:
- Old speed for Mbed OS cloning: 1 minute 12.19 seconds
- New speed for Mbed OS cloning: 26.57 seconds

Fixes #178
commit 87040323 (2021-03-03): project: Shallow clone repos

Without shallow cloning, cloning Mbed OS can take quite a long while.
Clone with a depth of 1 when resolving library dependencies in order to
speed up cloning. A downside of this approach is that the resulting
repository will not contain other git references than the initial one,
but this can be worked around by performing a subsequent fetch using
`git fetch`.

Note that shallow cloning is not always possible. When a lib file
contains a git hash instead of a branch or tag, it isn't currently
possible to shallow clone from GitHub. In such a case, we will convert a
 [... cut]
commit 8203b1e8 (2021-03-03): project: Shallow clone repos

Without shallow cloning, cloning Mbed OS can take quite a long while.
Clone with a depth of 1 when resolving library dependencies in order to
speed up cloning. A downside of this approach is that the resulting
repository will not contain other git references than the initial one,
but this can be worked around by performing a subsequent fetch using
`git fetch`.

Note that shallow cloning is not always possible. When a lib file
contains a git hash instead of a branch or tag, it isn't currently
possible to shallow clone from GitHub. In such a case, we will convert a
 [... cut]
closed 2021-03-05 by Patater