ARMmbed/mbed-tools #179: Function invoked with incorrect arguments for function named: mbed_set_linker_script

Azq2, 2021-02-06:
### Description

I have custom target with custom `CMakeFile.txt` https://github.com/Azq2/mbed-test-project/blob/master/TARGET_STM32F405XG/CMakeLists.txt

I overrided `CMakeLists.txt` for custom linker script and startup file.

But mbed build system don't load `TARGET_STM32F405XG/CMakeLists.txt` into cmake project and i see error:

```
CMake Error at mbed-os/CMakeLists.txt:136 (mbed_set_linker_script):
  mbed_set_linker_script Function invoked with incorrect arguments for
  function named: mbed_set_linker_script

-- Configuring incomplete, errors occurred!
```

But if i manually add loading of `TARGET_STM32F405XG/CMakeLists.txt` into main CMakeLists.txt of project:

```cmake
if(${MBED_TARGET} STREQUAL "STM32F405XG")
	include(${MBED_PATH}/tools/cmake/set_linker_script.cmake)
	add_subdirectory(TARGET_STM32F405XG)
endif()
```
All builds fine.

But i think, its not apropriate way. May be, mbed build system must automaticaly loads any `CMakeLists.txt` files in custom targets?

Steps for reproduce:
```sh
git clone https://github.com/Azq2/mbed-test-project
cd mbed-test-project
mbed-tools deploy
mbed-tools compile --toolchain GCC_ARM --mbed-target STM32F405XG
```

And we see error about mbed_set_linker_script

But if enable these lines (bad workaround): https://github.com/Azq2/mbed-test-project/blob/15f0021234df4b200f62a4f536fb7ed2004553f5/CMakeLists.txt#L12

No any errors occured.

### Issue request type

- [x] Enhancement
- [x] Bug

Patater, 2021-02-09:
Hi @Azq2 

We are currently updating documentation for how to add custom targets now that Mbed OS uses CMake. You can find [a draft PR for adding the documentation at mbed-os-5-docs](https://github.com/ARMmbed/mbed-os-5-docs/pull/1407/). It is indeed the case that your application's CMakeLists.txt would need to add the custom target directory, including the target-specific CMakeLists.txt file which tells CMake how to build for your target.

Patater, 2021-03-09:
We've merged https://github.com/ARMmbed/mbed-os-5-docs/pull/1407. The documentation will be published on the website shortly (with the next Mbed OS release).

closed 2021-03-09 by Patater