ARMmbed/mbed-tools #184: Adding option to explicitly specify build directory to "mbed-tools configure" / "mbed-tools compile" commands.

lukgni, 2021-02-10:
### Description
`mbed-tools configure` stores target specifc configuration (.mbedbuild/) in a project dir - this can't be overridden with different path.
It is a little bit inconvenient, especially in situation when we want to use single project directory to work with multiple HW targets (or with tests on multiple HW in CI/CD).

Maybe we could add a new option to `mbed-tool configure | mbed-tool compile` commands, which would explicitly specify build directory where .mbedbuild will be created (something similar to `-B` flag from cmake >3.19) 

For example:
```
mbed-tools configure -B <build_directory> ...
cmake  -B <build_directory> ...
cmake --build <build_directory>
```
or
```
mbed-tools compile -B <build_directory> ...
```

### Issue request type

- [X] Enhancement
- [ ] Bug

ladislas, 2021-02-11:
It's something I need as well. 

@Patater @rwalton-arm unless you guys are already working on that, I'm willing to make a PR. 

let me know. :)

Patater, 2021-02-11:
@ladislas We aren't working on this currently, but would appreciate a PR for it. Thanks!

boraozgen, 2021-03-22:
Any updates on this?