ARMmbed/mbed-tools #186: detect: External STLink support

jeromecoutant, 2021-02-11:
### Description

Hi

I am using an external st-link probe:
https://www.st.com/en/development-tools/stlink-v3set.html
https://www.st.com/en/development-tools/stlink-v3mini.html

Windows drive is well mounted, COM port is available,
but detect feature is not working:

````
$ mbedtools detect -a
ERROR: A problem occurred when looking up board data for connected devices.

More information may be available by using the command line option '-v'.
````

It is very useful for custom boards.

@MarceloSalazar 

### Issue request type

- [ ] Enhancement
- [x] Bug

MarceloSalazar, 2021-02-11:
@jeromecoutant can you confirm you have STLink attached to a supported (Mbed Enabled) board/MCU?

Is `mbedls` able to detect the board/MCU?

jeromecoutant, 2021-02-11:
> @jeromecoutant can you confirm you have STLink attached to a supported (Mbed Enabled) board/MCU?

Yes, STLink is attached to a board, but this board is of course not "mbed enabled" yet, as it is custom development board.

> Is `mbedls` able to detect the board/MCU?

No, as you know, it was working before, but not any more,
there is also the issue wih CLI1

Patater, 2021-02-19:
@jeromecoutant Could you please share the output of running `mbedtools -vvv detect -a`? There isn't enough information in this issue to understand why your particular board failed to be detected. Thanks!

jeromecoutant, 2021-02-19:
NB: mounted drive is E:

````
$ mbedtools -vvv detect -a
DEBUG: Failed getting an attribute value: <unknown>.AdditionalAvailability
DEBUG: Failed getting an attribute value: <unknown>.IdentifyingDescriptions
DEBUG: Failed getting an attribute value: <unknown>.MaxQuiesceTime
DEBUG: Failed getting an attribute value: <unknown>.OtherIdentifyingInfo
DEBUG: Failed getting an attribute value: <unknown>.PowerOnHours
DEBUG: Failed getting an attribute value: <unknown>.TotalPowerOnHours
DEBUG: Failed getting an attribute value: <unknown>.AdditionalAvailability
DEBUG: Failed getting an attribute value: <unknown>.IdentifyingDescriptions
DEBUG: Failed getting an attribute value: <unknown>.MaxQuiesceTime
DEBUG: Failed getting an attribute value: <unknown>.OtherIdentifyingInfo
DEBUG: Failed getting an attribute value: <unknown>.PowerOnHours
DEBUG: Failed getting an attribute value: <unknown>.TotalPowerOnHours
DEBUG: Failed getting an attribute value: <unknown>.AdditionalAvailability
DEBUG: Failed getting an attribute value: <unknown>.IdentifyingDescriptions
DEBUG: Failed getting an attribute value: <unknown>.MaxQuiesceTime
DEBUG: Failed getting an attribute value: <unknown>.OtherIdentifyingInfo
DEBUG: Failed getting an attribute value: <unknown>.PowerOnHours
DEBUG: Failed getting an attribute value: <unknown>.TotalPowerOnHours
DEBUG: Failed getting an attribute value: <unknown>.AdditionalAvailability
DEBUG: Failed getting an attribute value: <unknown>.IdentifyingDescriptions
DEBUG: Failed getting an attribute value: <unknown>.MaxQuiesceTime
DEBUG: Failed getting an attribute value: <unknown>.OtherIdentifyingInfo
DEBUG: Failed getting an attribute value: <unknown>.PowerOnHours
DEBUG: Failed getting an attribute value: <unknown>.TotalPowerOnHours
DEBUG: Failed getting an attribute value: <unknown>.AdditionalAvailability
DEBUG: Failed getting an attribute value: <unknown>.IdentifyingDescriptions
DEBUG: Failed getting an attribute value: <unknown>.MaxQuiesceTime
DEBUG: Failed getting an attribute value: <unknown>.OtherIdentifyingInfo
DEBUG: Failed getting an attribute value: <unknown>.PowerOnHours
DEBUG: Failed getting an attribute value: <unknown>.TotalPowerOnHours
DEBUG: Cannot retrieve the real name of volume Y:\. Reason: (87, 'GetVolumeNameForVolumeMountPoint', 'The parameter is incorrect.')
DEBUG: Attribute [DiskIndex] is undefined on this instance DiskPartition({}): 'DiskPartition' object has no attribute 'DiskIndex'
DEBUG: Attribute [PNPDeviceID] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'PNPDeviceID'
DEBUG: Attribute [SerialNumber] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'SerialNumber'
DEBUG: Attribute [DeviceID] is undefined on this instance DiskPartition({}): 'DiskPartition' object has no attribute 'DeviceID'
DEBUG: Attribute [Type] is undefined on this instance DiskPartition({}): 'DiskPartition' object has no attribute 'Type'
DEBUG: Attribute [Caption] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'Caption'
DEBUG: Attribute [DeviceID] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'DeviceID'
DEBUG: Attribute [Model] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'Model'
DEBUG: Attribute [InterfaceType] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'InterfaceType'
DEBUG: Attribute [MediaType] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'MediaType'
DEBUG: Attribute [Manufacturer] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'Manufacturer'
DEBUG: Attribute [SerialNumber] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'SerialNumber'
DEBUG: Attribute [Status] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'Status'
DEBUG: Attribute [PNPDeviceID] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'PNPDeviceID'
DEBUG: Cannot retrieve the real name of volume Y:\. Reason: (87, 'GetVolumeNameForVolumeMountPoint', 'The parameter is incorrect.')
DEBUG: Attribute [DiskIndex] is undefined on this instance DiskPartition({}): 'DiskPartition' object has no attribute 'DiskIndex'
DEBUG: Attribute [PNPDeviceID] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'PNPDeviceID'
DEBUG: Attribute [SerialNumber] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'SerialNumber'
DEBUG: Attribute [DeviceID] is undefined on this instance DiskPartition({}): 'DiskPartition' object has no attribute 'DeviceID'
DEBUG: Attribute [Type] is undefined on this instance DiskPartition({}): 'DiskPartition' object has no attribute 'Type'
DEBUG: Attribute [Caption] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'Caption'
DEBUG: Attribute [DeviceID] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'DeviceID'
DEBUG: Attribute [Model] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'Model'
DEBUG: Attribute [InterfaceType] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'InterfaceType'
DEBUG: Attribute [MediaType] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'MediaType'
[... cut ...]
DEBUG: Attribute [PNPDeviceID] is undefined on this instance DiskDrive({}): 'DiskDrive' object has no attribute 'PNPDeviceID'
INFO: Using the offline database to identify boards.
INFO: Unable to identify a board using the offline database, trying the online database.
DEBUG: Starting new HTTPS connection (1): os.mbed.com:443
WARNING: There was an error connecting to the online database. Please check your internet connection.
ERROR: A problem occurred when looking up board data for connected devices.

More information may be available by using the command line option '--traceback'.
````

jeromecoutant, 2021-02-19:
About

> Please check your internet connection.

ST network has proxy.
We can use http_proxy and https_proxy environment variable, used by curl and wget.

Patater, 2021-02-22:
I've raised https://github.com/ARMmbed/mbed-tools/issues/200 and https://github.com/ARMmbed/mbed-tools/issues/201 for two of the issues so far, but I don't think after fixing them we'll have completely gotten to the bottom of everything. We can at least fix what we've identified as issues so far and then try again afterwards to see how far we'll get.

Patater, 2021-03-05:
Hi @jeromecoutant. We've added some more verbose logging for when we attempt to connect to the net; hopefully this will help us see where the network access trouble is happening. We've also made network access optional for custom boards. We also verified we could use `http_proxy` to connect to the net from Windows, so we'd expect proxying to work.

Could you please try again with [mbed-tools 7.2.2.dev2 or later](https://pypi.org/project/mbed-tools/7.2.2.dev2/) and share back the new verbose log? Note that this is a pre-release and would need to be installed with `pip install -U --pre mbed-tools`. Thanks!

jeromecoutant, 2021-03-08:
Good!

```
$ mbedtools detect
ERROR: Unable to access the online database to identify a board.
No connected Mbed devices found.

$ mbedtools detect -a
ERROR: Unable to access the online database to identify a board.
Board name    Serial number             Serial port    Mount point(s)    Build target(s)
------------  ------------------------  -------------  ----------------  -----------------
<unknown>     0047004b3538511133363730  COM40          D:

```

Now, we just need the mock support :-)

Patater, 2021-03-09:
@jeromecoutant It looks like you still are having network issues. Could you please provide the output of `mbedtools -vvv detect -a` so that we can help debug?

jeromecoutant, 2021-03-09:
````
INFO: Using the offline database to identify boards.
INFO: Unable to identify a board using the offline database, trying the online database.
DEBUG: Starting new HTTPS connection (1): os.mbed.com:443
WARNING: Unable to verify an SSL certificate with requests.
WARNING: Unable to connect to the online database. Please check your internet connection.
ERROR: Unable to access the online database to identify a board.

````