ARMmbed/mbed-tools #200: Add support for network proxies

Patater, 2021-02-22:
### Description

Read from and use the `http_proxy` and `https_proxy` environment variables when attempting to obtain the online database (or any other network operations).

Some folks require proxies to access the net (as seen in https://github.com/ARMmbed/mbed-tools/issues/186), and we should respect that.

### Issue request type

- [X] Enhancement
- [ ] Bug

Patater, 2021-03-09:
We've added tests in https://github.com/ARMmbed/mbed-tools/pull/218 to ensure we work with proxies via the `https_proxy` and `http_proxy` environment variables. We've also added more verbose logging (via https://github.com/ARMmbed/mbed-tools/pull/213) so we can tell where the network issues are if any trouble comes up in the future.

jeromecoutant, 2021-03-09:
ST network needs to use https_proxy and http_proxy environment variables.

````
$ mbedtools --version
7.2.2.dev2

$ mbedtools -vvv detect -a
...
INFO: Using the offline database to identify boards.
INFO: Unable to identify a board using the offline database, trying the online database.
DEBUG: Starting new HTTPS connection (1): os.mbed.com:443
WARNING: Unable to verify an SSL certificate with requests.
WARNING: Unable to connect to the online database. Please check your internet connection.
ERROR: Unable to access the online database to identify a board.
...
````

Patater, 2021-03-09:
@jeromecoutant Looks like there is an issue with the certificate for os.mbed.com on your network. We currently work with `https_proxy` and `https_proxy`, but do we also need to support an alternative root certificate for SSL on your network? That's probably a separate issue we need to work on with you. I've attached screenshots of the certicate as seen from my web browser. It'd be interesting to compare with what you see from your browser, if at all different.

[image]
[image]

Patater, 2021-03-09:
I'll close this one as I've opened a new issue about the PKI issues: https://github.com/ARMmbed/mbed-tools/issues/221

closed 2021-03-09 by Patater