ARMmbed/mbed-tools #211: CMake Error:  could not find load file /.mbedbuild/mbed_config.cmake

cyliangtw, 2021-03-03:
### Description

In mbed-tools v5.2.0, it could work on mbed OS v6.7 and generated `/.mbedbuild/` folder.
In mbed-tools v7.2.0, suffer `could not find load file /.mbedbuild/mbed_config.cmake` issue on both of mbed OS v6.7 and last mbed OS master branch.

Error message as below:
```
[cygwin~mbed-os-example-blinky-cmake]$ mbed-tools compile -m NUMAKER_IOT_M487 -t GCC_ARM
Configuring project and generating build system...
CMake Error at mbed-os/tools/cmake/app.cmake:10 (include):
  include could not find load file:

    C:/ARM_mbed/examples/mbed-os-example-blinky-cmake/.mbedbuild/mbed_config.cmake
Call Stack (most recent call first):
  CMakeLists.txt:10 (include)

```

### Issue request type

- [ ] Enhancement
- [x] Bug

0xc0170, 2021-03-03:
This include is failing `include(${MBED_CONFIG_PATH}/mbed_config.cmake)`

Can you share what's the value of `MBED_CONFIG_PATH` ? This is cached variable and it should follow the pattern `mbed-os-example-blinky/cmake_build/DISCO_L475VG_IOT01A/develop/GCC_ARM` or similar (starting from mbed-tools 7.x).

I am currently working with 7.0.0 locally and latest master and haven't hit this issue with the latest examples (development branch). What's the version of the example you are building (have you updated?) ?

cyliangtw, 2021-03-04:
Thanks. 
After git pull `mbed-os-example-blinky`, `MBED_CONFIG_PATH` is updated and mbed-tools 7.2.0 could work well with OS v6.8 .
However, if I apply the last Mbed OS master branch, cmake build encounter fatal error "`can't find core_m4.h`", it seems `mbed-os/cmsis/CMSIS_5/CMSIS/TARGET_CORTEX_M/Include/` not add into include path. 
```
[cygwin~mbed-os-example-blinky-cmake]$ mbed-tools compile -m NUMAKER_IOT_M487 -t GCC_ARM
Configuring project and generating build system...
..........................................................
FAILED: CMakeFiles/mbed-os-example-blinky.dir/mbed-os/drivers/source/DigitalIn.cpp.obj
C:\PROGRA~2\GNUARM~1\92020-~1\bin\AR10B2~1.EXE @CMakeFiles\mbed-os-example-blinky.dir\mbed-os\drivers\source\DigitalIn.cpp.obj.rsp -MD -MT CMakeFiles/mbed-os-example-blinky.dir/mbed-os/drivers/source/DigitalIn.cpp.obj -MF CMakeFiles\mbed-os-example-blinky.dir\mbed-os\drivers\source\DigitalIn.cpp.obj.d -o CMakeFiles/mbed-os-example-blinky.dir/mbed-os/drivers/source/DigitalIn.cpp.obj -c ../../../../mbed-os/drivers/source/DigitalIn.cpp
In file included from ../../../../mbed-os/targets/TARGET_NUVOTON/TARGET_M480/device/cmsis.h:22,
                 from ../../../../mbed-os/targets/TARGET_NUVOTON/TARGET_M480/./objects.h:22,
                 from ../../../../mbed-os/targets/TARGET_NUVOTON/TARGET_M480/./device.h:24,
                 from ../../../../mbed-os/platform/include/platform/platform.h:28,
                 from ../../../../mbed-os/drivers/./include/drivers/DigitalIn.h:20,
                 from ../../../../mbed-os/drivers/source/DigitalIn.cpp:18:
../../../../mbed-os/targets/TARGET_NUVOTON/TARGET_M480/device/M480.h:203:10: fatal error: core_cm4.h: No such file or directory
  203 | #include "core_cm4.h"               /* Cortex-M4 processor and core peripherals 
```

cyliangtw, 2021-03-04:
I fixed the above issue "`fatal error: core_cm4.h: No such file`" in https://github.com/ARMmbed/mbed-os/pull/14376.

closed 2021-03-04 by cyliangtw

0xc0170, 2021-03-04:
Thank you for fixing this.