ARMmbed/mbed-tools #212: Mbed tools can't compile an application if mbed-os.lib isn't present

pan-, 2021-03-03:
### Description

I have a repository that contains several mbed-os applications. To avoid unnecessary duplicates of mbed-os, I use a single clone of mbed-os and symlink it inside each of the mbed application. 

`mbed-tools compile` refuses to start the compilation if `mbed-os.lib` is not in the application folder. If I add an empty `mbed-os.lib` file then the compilation proceed.

Is it the expected behaviour ? 

### Issue request type

- [ ] Enhancement
- [ ] Bug
- [x] Question

Patater, 2021-03-09:
We support this use case via the `--mbed-os-path` option, where you can explicitly specify where to use Mbed OS from. Use that instead of adding a blank `mbed-os.lib` file which is weird and may not do what you expect in the future (or now). Both `compile` and `configure` support the `--mbed-os-path` option.

closed 2021-03-10 by Patater

jeromecoutant, 2021-03-10:
I don't understand the answer...
I agree with @pan- tools doesn't need mbed-os.lib file at all, so why refusing the compilation...