ARMmbed/mbed-tools #214: Cannot build EP_ATLAS Mbed target

hugueskamba, 2021-03-04:
### Description

Run:
```
$ rm -rf cmake_build && mbedtools compile -t ARM -m EP_ATLAS
```

Expected output:
The build succeeds.

Actual output:
```
Configuring project and generating build system...
Traceback (most recent call last):
  File "/usr/local/bin/mbedtools", line 8, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/site-packages/mbed_tools/cli/main.py", line 38, in invoke
    super().invoke(context)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/mbed_tools/cli/build.py", line 99, in build
    generate_config(mbed_target.upper(), toolchain, program)
  File "/usr/local/lib/python3.9/site-packages/mbed_tools/build/config.py", line 34, in generate_config
    config = assemble_config(target_build_attributes, program.root, program.files.app_config_file)
  File "/usr/local/lib/python3.9/site-packages/mbed_tools/build/_internal/config/assemble_build_config.py", line 29, in assemble_config
    return _assemble_config_from_sources(target_attributes, mbed_lib_files, mbed_app_file)
  File "/usr/local/lib/python3.9/site-packages/mbed_tools/build/_internal/config/assemble_build_config.py", line 51, in _assemble_config_from_sources
    config.update(source.from_file(config_file, target_filters=current_filter_data.labels))
  File "/usr/local/Cellar/python@3.9/3.9.1_6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/_collections_abc.py", line 856, in update
    self[key] = other[key]
  File "/usr/local/lib/python3.9/site-packages/mbed_tools/build/_internal/config/config.py", line 29, in __setitem__
    self._handle_overrides(item)
  File "/usr/local/lib/python3.9/site-packages/mbed_tools/build/_internal/config/config.py", line 46, in _handle_overrides
    setting = self._find_config_setting(lambda x: x.name == override.name and x.namespace == override.namespace)
  File "/usr/local/lib/python3.9/site-packages/mbed_tools/build/_internal/config/config.py", line 76, in _find_config_setting
    raise ValueError("Could not find element.")
ValueError: Could not find element.
```
### Issue request type

- [ ] Enhancement
- [x] Bug

rwalton-arm, 2021-03-05:
We changed this message to a warning and added more verbose logging in 7.2.1, please retry with that version (it should only warn instead of failing with a ValueError) and then fix the invalid overrides in your config.

hugueskamba, 2021-03-05:
It builds with:
```
$ mbedtools --version
7.2.1
```

closed 2021-03-05 by hugueskamba

Patater, 2021-03-05:
I tried and got a new error.

```
> mbedtools compile -t ARM -m EP_ATLAS
Configuring project and generating build system...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is ARMClang
-- Found assembler: /Users/jaeame01/mbed-studio-tools/ac6/bin/armclang
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /Users/jaeame01/mbed-studio-tools/ac6/bin/armclang
-- Check for working C compiler: /Users/jaeame01/mbed-studio-tools/ac6/bin/armclang - broken
CMake Error at /Applications/CMake.app/Contents/share/cmake-3.19/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

    "/Users/jaeame01/mbed-studio-tools/ac6/bin/armclang"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /tmp/mbed-os-example-blinky/cmake_build/EP_ATLAS/develop/ARM/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/local/bin/ninja cmTC_977ba && [1/2] Building C object CMakeFiles/cmTC_977ba.dir/testCCompiler.c.obj
    FAILED: CMakeFiles/cmTC_977ba.dir/testCCompiler.c.obj 
    /Users/jaeame01/mbed-studio-tools/ac6/bin/armclang   -mthumb -Wno-armcc-pragma-push-pop -Wno-armcc-pragma-anon-unions -Wno-reserved-user-defined-literal -Wno-deprecated-register -fdata-sections -fno-exceptions -fshort-enums -fshort-wchar -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -o CMakeFiles/cmTC_977ba.dir/testCCompiler.c.obj -c testCCompiler.c
    armclang: error: This variant of Arm Compiler is licensed only for use with Mbed Studio
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  mbed-os/tools/cmake/app.cmake:21 (enable_language)
  CMakeLists.txt:10 (include)

-- Configuring incomplete, errors occurred!
See also "/tmp/mbed-os-example-blinky/cmake_build/EP_ATLAS/develop/ARM/CMakeFiles/CMakeOutput.log".
See also "/tmp/mbed-os-example-blinky/cmake_build/EP_ATLAS/develop/ARM/CMakeFiles/CMakeError.log".
ERROR: CMake invocation failed!

More information may be available by using the command line option '-v'.
```

Looks like mbed-os doesn't set [`CMAKE_TRY_COMPILE_TARGET_TYPE` to `STATIC_LIBRARY` ](https://cmake.org/cmake/help/latest/variable/CMAKE_TRY_COMPILE_TARGET_TYPE.html) maybe?

Patater, 2021-03-05:
Ah, nevermind. I didn't catch that your reproduction instructions used the `ARM` toolchain. I only have `GCC_ARM` here on my Mac.

Patater, 2021-03-05:
With `GCC_ARM`:
```
[198/223] Building CXX object CMakeFiles/mbed-os...2840/TARGET_EP_ATLAS/ONBOARD_TELIT_ME310.cpp.obj
FAILED: CMakeFiles/mbed-os-example-blinky.dir/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_EP_ATLAS/ONBOARD_TELIT_ME310.cpp.obj 
/usr/local/bin/ccache /Users/jaeame01/.local/bin/arm-none-eabi-g++ @CMakeFiles/mbed-os-example-blinky.dir/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_EP_ATLAS/ONBOARD_TELIT_ME310.cpp.obj.rsp -MD -MT CMakeFiles/mbed-os-example-blinky.dir/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_EP_ATLAS/ONBOARD_TELIT_ME310.cpp.obj -MF CMakeFiles/mbed-os-example-blinky.dir/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_EP_ATLAS/ONBOARD_TELIT_ME310.cpp.obj.d -o CMakeFiles/mbed-os-example-blinky.dir/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_EP_ATLAS/ONBOARD_TELIT_ME310.cpp.obj -c ../../../../mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_EP_ATLAS/ONBOARD_TELIT_ME310.cpp
In file included from ../../../../mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_EP_ATLAS/ONBOARD_TELIT_ME310.cpp:22:
../../../../mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_EP_ATLAS/ONBOARD_TELIT_ME310.h:21:10: fatal error: TELIT_ME310.h: No such file or directory
   21 | #include "TELIT_ME310.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
[199/223] Building CXX object CMakeFiles/mbed-os...le-blinky.dir/mbed-os/rtos/source/Thread.cpp.obj
ninja: build stopped: subcommand failed.
ERROR: CMake invocation failed!

More information may be available by using the command line option '-v'.
```

hugueskamba, 2021-03-05:
> With `GCC_ARM`:
>
> ```
> [198/223] Building CXX object CMakeFiles/mbed-os...2840/TARGET_EP_ATLAS/ONBOARD_TELIT_ME310.cpp.obj
> FAILED: [...]

We have the same issue with the ARM toolchain. We are currently refactoring connectivity drivers and will add the TELIT driver to this Mbed board as a dependency.
FYI @0xc0170