ARMmbed/mbed-tools #215: Unable to flash mocked board

sstelian, 2021-03-04:
### Description

I'm trying to flash a custom board based on the NUCLEO_H743ZI board using a NUCLEO_L476RG as a STLink. Everything works with mbed-cli, but not with mbed-tools. I moked the platform using mbedls --mock 0799:NUCLEO_H743ZI2
. Mbed-tools compiles the code, but does not flash it. The error is: 

> ERROR: Could not identify a board with the product code: '0799'.
ERROR: No Mbed enabled devices found.

I figured out that mbed-tools does not detect the mocked board. On the other hand, mbed-cli does detect it:

> [stelian@localhost anvil-v4]$ mbed detect
[mbed] Working path "/storage/work/anvil-v4" (program)
[mbed] WARNING: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: cbor
       You can install all missing modules by running "pip install -r requirements.txt" in "/storage/work/anvil-v4/mbed-os"
       On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
[mbed] Detected NUCLEO_H743ZI2, port /dev/ttyACM0, mounted /run/media/stelian/NODE_L476RG, interface version V2J28M17:
[mbed] Supported toolchains for NUCLEO_H743ZI2
| Target         | mbed OS 2 | mbed OS 5 |    ARM    | uARM |  GCC_ARM  |    IAR    |
|----------------|-----------|-----------|-----------|------|-----------|-----------|
| NUCLEO_H743ZI2 |     -     | Supported | Supported |  -   | Supported | Supported |
Supported targets: 1
Supported toolchains: 3
[stelian@localhost anvil-v4]$ mbed-tools detect
ERROR: Could not identify a board with the product code: '0799'.
No connected Mbed devices found.
 

### Issue request type

- [ ] Enhancement
- [ x] Bug

I would expect the same behavior as mbed-cli and I would expect the mocking functionality to work with mbed-tools as well.

Patater, 2021-03-11:
Thanks, @sstelian. We had a think about what it'd take to support this and have planned it. We'll update the tasks as we learn and go.

rwalton-arm, 2021-07-29:
The task list was out of date. We're not planning on pulling mbedls code into this project. We need to find an alternative solution. I've updated the task list to reflect this.

jeromecoutant, 2021-09-30:
Any update ?
Any solution to be able to use custom boards (with no device id) with CLI2 ?

ladislas, 2021-09-30:
We are using cmake (not CLI2 though) with a custom board (STM32F769) and our solution was to simply use openocd with the following commands:

```sh
# flash the board
openocd -f interface/stlink.cfg -c 'transport select hla_swd' -f target/stm32f7x.cfg -c 'program $(BIN_PATH) 0x08000000' -c exit

# wait for 1 second
sleep1

# reset the board
openocd -f interface/stlink.cfg -c 'transport select hla_swd' -f target/stm32f7x.cfg -c init -c 'reset run' -c exit
```

We have that in a script but it could be made a custom target in cmake I guess.

ladislas, 2021-09-30:
`-f`, `--flash` or simply `mbed-tools flash` should be the default by copy/paste as you said.

For custom boards though, I think if would be nice to let people provide the script they need with the tool they need. Custom boards by definition are all different so the settings might be different. The tool cannot now that, but the people using the tool should be able to provide a script or implement a cmake target to call `mbed-tools flash --custom-board`.

I don't think it's the job of the tool to make that script or try to automate the generation of the script.