ARMmbed/mbed-tools #217: Compile error while using custom Lora radio in custom board.

kvkc97, 2021-03-05:
Hi,

I am compiling the mbed lorawan example code using MBED CLI2. I have added a custom target board for STM3WB55RG. I have to use a custom Lora radio driver based on LR1110. I added the LR1110 driver in the folder mbed-os\connectivity\drivers\lora as a component like (SX1276) and created the CMake file. I also modified the custom_target file (to add LR1110 as component) using "components_add:" command. Also modified lora_radio_helper.h and CMake file at mbed-os\connectivity\drivers\lora folder. However, while compiling below error is generated. The same error is also generated when STM32WL is used as the component. STM32WL is defined in the mbed-os libraray and I didnot make any changes to it.
[image]

I have attached the CMake files which were modified and also the snaps of the lora_radio_helper.h and custom_target.json where modification were made.
[CMakeLists.txt](https://github.com/ARMmbed/mbed-tools/files/6088750/CMakeLists.txt)
[CMakeLists.txt](https://github.com/ARMmbed/mbed-tools/files/6088753/CMakeLists.txt)
[image]
[image]

Regards
Keerthi

Patater, 2021-03-09:
Hi Keerthi,

It shouldn't be necessary to modify `mbed-os` itself to add your driver or custom target. Could you add the driver to your application's folder instead? You'll need to add the driver's folder (`LR1110`) via `add_subdirectory(LR1110)` in your application's `CMakeLists.txt`. It shouldn't be necessary to use `components_add` json to include your driver.

Here is the file structure I'd expect for a custom target and custom driver. (Stuff with a `/` at the end would be a directory.)

```
CMakeLists.txt
custom_targets.json
main.cpp
mbed_app.json
mbed-os/
mbed-os.lib
TARGET_STM3WB55RG/
LR1110/
```

LR1110 would need a CMakeLists.txt in it, with at least `target_include_directories()` and maybe also `target_sources()` It might define a new library (maybe `lr1110`) for your application to link with. You'd link to such a library from your application's `CMakeLists.txt` with:
```target_link_libraries(${APP_TARGET}
    PRIVATE
        mbed-os
        mbed-lorawan
        lr1110
)```

It's not published yet (will be soon), but we have [custom target documentation available in our docs repo](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/porting/custom-target-porting.md) which you might find helpful.

Hope this helps.

kvkc97, 2021-03-16:
I added the driver's folder to the application's folder. The file structure of the application is as shown in the above reply.  However, there are some doubts regarding the CMakeLists.txt file in the LR1110 folder. Can you share a general format to create the CMakeLists file? Also, should we generate the .lib file for LR1110, if so how to proceed with that?  
I checked the above mentioned documentation but it has less details on creating the CMakeLists file for library.

kvkc97, 2021-03-24:
Can I get an update on this query..

Patater, 2021-03-26:
@kvkc97 You don't need a .lib for LR1110, assuming it's just a folder in your application's folder. The CMakeLists.txt should roughly look similar to the custom target one, although with any target inheritance. See our custom target CMakeLists.txt here for reference: https://github.com/ARMmbed/mbed-tools/blob/master/travis-ci/test-data/TARGET_IMAGINARYBOARD/CMakeLists.txt

I agree the documentation for how to make libraries for use with your application is lacking. This is a general (non-Mbed) example on how to structure a library, with an application, in a single project: https://gitlab.com/CLIUtils/modern-cmake/-/tree/master/examples/extended-project (from the excellent guide https://cliutils.gitlab.io/modern-cmake/)

kvkc97, 2021-03-27:
I modified the LR110/CMakeLists.txt file as above. But it generates CMake error. I have attached the lr1110 folder structure, CMakeLists.txt file and the CMakeError log. Could you suggest what might be the issue?
Given below is the error displayed while compiling.
CMake Error at lr1110/CMakeLists.txt:6 (target_include_directories):
  Cannot specify include directories for target "lr1110" which is not built
  by this project.
[image]
[image]
[CMakeLists.txt](https://github.com/ARMmbed/mbed-tools/files/6215864/CMakeLists.txt)
[CMakeError.log](https://github.com/ARMmbed/mbed-tools/files/6215863/CMakeError.log)

Patater, 2021-04-08:
@kvkc97 In your `lr1110/CMakeLists.txt`, could you try moving `add_library()` up to before `target_include_directories()`?

lr1110/CMakeLists.txt
```
# Copyright (c) 2020 ARM Limited. All rights reserved.
# SPDX-License-Identifier: Apache-2.0

#add_subdirectory(src)

add_library(lr1110
    PUBLIC
        lr1110_bootloader.c
        lr1110_crypto_engine.c
		lr1110_driver_version.c
		lr1110_gnss.c 
		lr1110_hal.c 
		lr1110_radio.c 
		lr1110_regmem.c 
		lr1110_system.c 
		lr1110_wifi.c 
		system.c 
		system_gpio.c 
		system_spi.c 
		system_time.c 
)

target_include_directories(lr1110
    PUBLIC
        include
)
```

Also, could you share your mbed_app.json? Are you trying to build "Mbed OS baremetal"?

kvkc97, 2021-04-08:
Hi,

I modified the lr1110/CMakeLists.txt as mentioned. The error obtained is attached. 
[image]

mbed_app.json file is also attached below.
[mbed_app.txt](https://github.com/ARMmbed/mbed-tools/files/6280211/mbed_app.txt)
No I am not building Mbed OS baremetal.

Patater, 2021-04-08:
Ah, sorry. Drop the "PUBLIC" from the `add_library()` invokation.

kvkc97, 2021-04-09:
I got the attached error.
[image]

jeromecoutant, 2021-04-09:
STM32WL supports LORA by default, so it should compile and work.
For STM32WB you need an external radio, and then add a COMPONENT in your local json configuration.

kvkc97, 2021-04-09:
I want to use lr1110 as the lora driver. I added the below statement in custom target json file. This gave the same error again. Is there any other step required?
"components_add": ["lr1110"],
How to create the external radio for lr1110 in the code?

jeromecoutant, 2021-04-09:
See there:
https://github.com/ARMmbed/mbed-os/tree/master/connectivity/drivers/lora

kvkc97, 2021-04-13:
I tried to create external radio for lr1110 by modifying the lora_radio_helper.cpp file as follows. 
[image]
I also declared the class constructor in lr1110_radio.h file as below.
[image]
I got the following compilor error.
[image]

Patater, 2021-04-15:
This is beyond me at this point, sorry. It appears the build system is working well, but there may be some LoRa-specific configuration or implementation missing. We have [a guide for porting LoRa](https://os.mbed.com/docs/mbed-os/v6.9/porting/lora-port.html) which you might find helpful.