ARMmbed/mbed-tools #221: Support alternative certificate stores

Patater, 2021-03-09:
### Description

Not confirmed this is actually the issue, but it appears some of our users need a way to use a non-default certificate store for PKI. At the very least, we know that authenticating os.mbed.com over https is failing for them.

### Issue request type

- [ ] Enhancement
- [X] Bug

Patater, 2021-03-09:
From @jeromecoutant in https://github.com/ARMmbed/mbed-tools/issues/200#issuecomment-794109730
```
$ mbedtools --version
7.2.2.dev2

$ mbedtools -vvv detect -a
...
INFO: Using the offline database to identify boards.
INFO: Unable to identify a board using the offline database, trying the online database.
DEBUG: Starting new HTTPS connection (1): os.mbed.com:443
WARNING: Unable to verify an SSL certificate with requests.
WARNING: Unable to connect to the online database. Please check your internet connection.
ERROR: Unable to access the online database to identify a board.
...
```

jeromecoutant, 2021-09-24:
Proposition:
in https://github.com/ARMmbed/mbed-tools/blob/master/src/mbed_tools/cli/list_connected_devices.py
Add a new option "--nossl"

And in https://github.com/ARMmbed/mbed-tools/blob/master/src/mbed_tools/targets/_internal/board_database.py#L107

Change to `return requests.get(_BOARD_API, headers=header, verify=SSL_OPTION)`
- SSL_OPTION is True by default
- SSL_OPTION is False if nossl is requested by user