ARMmbed/mbed-tools #234: Does not compile with --mbed-os-path option

boraozgen, 2021-03-18:
**Describe the bug**
As in title.

**To Reproduce**
Steps to reproduce the behavior:
1. git init
2. git submodule init
3. git submodule add https://github.com/ARMmbed/mbed-os.git
4. mbed-tools configure -m K64F -t GCC_ARM --mbed-os-path mbed-os
5. mbed-tools compile -m K64F -t GCC_ARM --mbed-os-path mbed-os

Error:
`ValueError: Setting storage_filesystem.rbp_internal_size already defined. You cannot duplicate config settings!`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 10

**Mbed (please complete the following information):**
 - Device: any
 - Mbed OS Version: 6.9.0
 - Mbed CLI 2 Version: 7.3.3

**Additional context**
Add any other context about the problem here.

boraozgen, 2021-03-18:
Seems to be a problem with my python dependencies. Seems to be working now. Closing.

boraozgen, 2021-04-08:
This keeps happening for me. Reopening.

Patater, 2021-04-08:
@boraozgen What did you find the issue with your Python dependencies was?

boraozgen, 2021-04-08:
Well I am not that experienced with python&pip so I removed and reinstalled some stuff and it started working. Now it seems broken again. I cannot point to the change which broke/fixed it. I updated the issue description with my current dependency tree.

Patater, 2021-04-08:
I can confirm I am able to reproduce this. Looks we are searching the same settings twice. We'll have to avoid this.

Patater, 2021-04-08:
I'm also able to reproduce with `mbed-tools compile -m K64F -t GCC_ARM --mbed-os-path extern/mbed-os`. Probably any path to mbed-os within the application directory would hit this issue.

ladislas, 2021-05-01:
I'm also hitting the same issue! :)

PR #285 (2021-05-26): config: Ensure duplicate config file paths are discarded
### Description

When `mbed-os-path` was given on the command line, and it happened to be
a subdirectory of the project path it would be searched twice for
mbed_lib.json files. This would cause duplicate config parameter
exceptions to be raised.

Ensure duplicated config file paths are discarded by resolving the
absolute paths. This is so `Path.__eq__` can detect duplicate entries and
they are filtered when building the final set. Also ensure we handle the
case where a subdirectory of the search path is a symlink.

Fixes #234

### Test C [... cut]
commit 3958bdc7 (2021-05-26): config: Ensure duplicate config file paths are discarded

When `mbed-os-path` was given on the command line, and it happened to be
a subdirectory of the project path it would be searched twice for
mbed_lib.json files. This would cause duplicate config parameter
exceptions to be raised.

Ensure duplicated config file paths are discarded by resolving the
absolute paths. This is so Path.__eq__ can detect duplicate entries and
they are filtered when building the final set. Also ensure we handle the
case where a subdirectory of the search path is a symlink.

Fixes #234
closed 2021-05-26 by Patater