ARMmbed/mbed-tools #238: Missing instructions on how to tell Cmake where to find your compiler

maclobdell, 2021-03-22:
**Describe the bug**

The instructions say to have a compiler installed, which I do (ARM GCC 9). I have also installed CMake and added to my path. 

I’m on Mac OS X Big Sur.

Cmake is installed here:
	/Applications/CMake.app/Contents/bin 

GCC for ARM is installed here:
	/Applications/ARM/bin

In the prior generation of Mbed tools I set my compiler path with `mbed config -G GCC_ARM_PATH=/Applications/ARM/bin`. However, I don't now how to do that in the new tools.

If I blindly follow the steps in the guide, I get the following error. It appears that Cmake hasn’t been told where the correct compiler is.  It is trying to use the Apple version. 

```
-- The C compiler identification is AppleClang 11.0.3.11030032
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- The ASM compiler identification is Clang
-- Found assembler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - broken
-- Configuring incomplete, errors occurred!
See also "/Users/maclob01/Documents/mbed-tools-2/mbed-os-example-blinky/cmake_build/CMakeFiles/CMakeOutput.log".
See also "/Users/maclob01/Documents/mbed-tools-2/mbed-os-example-blinky/cmake_build/CMakeFiles/CMakeError.log".
```

**To Reproduce**
Steps to reproduce the behavior:

1. Follow the guide on a Mac without having anything previously setup
https://os.mbed.com/docs/mbed-os/v6.8/build-tools/use.html

**Expected behavior**
Clear information about how to configure cmake to use the right compiler.

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - OS: Mac
 - Version: 11.11

**Mbed (please complete the following information):**
 - Device: n/a
 - Mbed OS Version: n/a
 - Mbed CLI 2 Version: 7.4.0

**Additional context**
n/a

ladislas, 2021-03-22:
I would highly recommend you use homebrew (https://brew.sh) to install cmake and arm-gcc. 

Are you on a intel or M1 Mac?

ladislas, 2021-03-22:
You can use the following for arm-gcc: https://github.com/osx-cross/homebrew-arm

For cmake juste `brew install cmake` will do.

0xc0170, 2021-03-23:
I assume this is related to https://github.com/ARMmbed/mbed-os/issues/14446 (regarding CMAKE_<LANG>_COMPILER set up. We hardcoded it and it should be fixed as CMake provides a way to configure it.

Patater, 2021-03-23:
@maclobdell Is `arm-none-eabi-gcc` in your path? What's the result of running `which arm-none-eabi-gcc`?

maclobdell, 2021-03-23:
Thanks for the help. Here are my responses.
- I have an Intel Mac. Recently upgraded to Big Sur.
- The compiler wasn't part of my path. I tried adding it. Still no luck.
- I cleaned up my brew installation (issue with Big Sur)
- I reinstalled cmake and the compiler using brew
- removed the cmake_build folder. started over...

unfortunately, I'm back to the same problem.

boraozgen, 2021-03-23:
The documentation is outdated. The build folder is not `cmake_build` anymore, but contains the target and the compiler options given to `mbed-tools configure`. Check your `cmake_build` directory to find where `mbed_config.cmake` is located and provide that path to `-B`.

Furthermore the current system does not allow providing C_C(XX)_COMPILER flags. You just have to add your compiler path to PATH. This is an open issue: https://github.com/ARMmbed/mbed-os/issues/14446

maclobdell, 2021-03-23:
interesting. this command worked. 

```
cmake -S . -B cmake_build/K64F/develop/GCC_ARM -GNinja -DCMAKE_BUILD_TYPE=debug
```

Awesome! 

I definitely recommend updating the documentation to clarify this.  Also, it would be good to mention how the tools select the compiler from the path and you might run into trouble if you have multiple compilers installed.

maclobdell, 2021-04-22:
I left this open because I do think it would be good to update the documentation to clarify things. I will leave it up to the maintainers to decide to close it or not.

Patater, 2021-04-23:
Hi @maclobdell We've updated the "How to use CMake" docs at https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/tools/mbed_cli_2/use.md#build-the-project-with-cmake-advanced

Layty, 2024-01-10:
@Patater  I have two gcc like follow, so how can I focus the second , I have  used 
-D CMAKE_C_COMPILER=xxx
-D CMAKE_CXX_COMPILER=xxx

C:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q2-update\bin\arm-none-eabi-g++.exe
C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-g++.exe