ARMmbed/mbed-tools #243: Definitions in mbed_lib.json are not generated for target link libraries

harmut01, 2021-03-25:
**Describe the bug**
When adding a target library via target_link_library, the tool does not generate the necessary definitions from the libraries 'mbed_lib.json`.

**To Reproduce**
Steps to reproduce the behavior:
1. Add library (i.e. mbed-events) as a target link library
2. Configure or compile application: `mbedtools configure -t ARM -m RZ_A1H`
3. Definitions for library can't be found `mbed_config.cmake`

**Expected behavior**
Tool should automatically generate macros from that linked libraries mbed_lib.json configs .i.e. MBED_CONF_EVENTS_SHARED_EVENTSIZE=2048 for the following: 
``` json       
       "shared-stacksize": {
            "help": "Stack size (bytes) for shared event queue thread",
            "value": 2048
        }
```
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [MacOS]
 - Version: [11.2.1]

**Mbed (please complete the following information):**
 - Device: [K64F]
 - Mbed OS Version: [6.9.0]
 - Mbed CLI 2 Version: [7.5.0]

**Additional context**
Add any other context about the problem here.

rwalton-arm, 2021-03-26:
I've just followed the repro steps and I ended up with the following in my `mbed_config.cmake`

```
  141     "-DMBED_CONF_EVENTS_SHARED_STACKSIZE=2048"
  142     "-DMBED_CONF_EVENTS_SHARED_EVENTSIZE=768"
  143     "-DMBED_CONF_EVENTS_SHARED_DISPATCH_FROM_APPLICATION=0"
  144     "-DMBED_CONF_EVENTS_SHARED_HIGHPRIO_STACKSIZE=1024"
  145     "-DMBED_CONF_EVENTS_SHARED_HIGHPRIO_EVENTSIZE=256"
  146     "-DMBED_CONF_EVENTS_USE_LOWPOWER_TIMER_TICKER=0"
```

Am I missing something? Could you post a verbose log from the tool?

harmut01, 2021-03-30:
I should note I encountered the above using the blinky-baremetal example, and the target library was added to the applications CMakeLists.txt. Here is the verbose logging output: 
```log
DEBUG: Searching for mbed-os.lib file at path /Users/harmut01/Projects/mbed-examples/blinky-baremetal
DEBUG: mbed-os.lib file found at /Users/harmut01/Projects/mbed-examples/blinky-baremetal
INFO: Found existing Mbed program at path '/Users/harmut01/Projects/mbed-examples/blinky-baremetal'
DEBUG: Loading JSON file /Users/harmut01/Projects/mbed-examples/blinky-baremetal/mbed-os/targets/targets.json
DEBUG: Loading JSON file /usr/local/lib/python3.9/site-packages/mbed_tools/targets/_internal/data/targets_metadata.json
DEBUG: Extracting config setting 'console-uart'='{'help': 'Target has UART console on pins CONSOLE_TX, CONSOLE_RX. Value is only significant if target has SERIAL device.', 'value': True}'
DEBUG: Extracting config setting 'console-uart-flow-control'='{'help': 'Console hardware flow control. Options: null, RTS, CTS, RTSCTS.', 'value': None}'
DEBUG: Extracting config setting 'network-default-interface-type'='{'help': 'Default network interface type. Typical options: null, ETHERNET, WIFI, CELLULAR, MESH', 'value': 'ETHERNET'}'
DEBUG: Extracting config setting 'deep-sleep-latency'='{'help': 'Time in ms required to go to and wake up from deep sleep (max 10)', 'value': 0}'
DEBUG: Extracting config setting 'app_offset'='{'help': 'Application start offset in ROM', 'value': None}'
DEBUG: Extracting config setting 'header_offset'='{'help': 'Application header offset in ROM', 'value': None}'
DEBUG: Extracting config setting 'boot-stack-size'='{'help': 'Define the boot stack size in bytes. This value must be a multiple of 8', 'value': '0x1000'}'
DEBUG: Extracting config setting 'mpu-rom-end'='{'help': 'Last address of ROM protected by the MPU', 'value': '0x0fffffff'}'
DEBUG: Extracting config setting 'default-form-factor'='{'help': "Default form factor of this board taken from supported_form_factors. This must be a lowercase string such as 'arduino'", 'value': None}'
DEBUG: Extracting config setting 'tickless-from-us-ticker'='{'help': 'Run tickless from the microsecond ticker rather than the low power ticker. Running tickless off of the microsecond ticker improves interrupt latency on targets which use lpticker_delay_ticks', 'value': False}'
DEBUG: Extracting config setting 'init-us-ticker-at-boot'='{'help': 'Initialize the microsecond ticker at boot rather than on first use, and leave it initialized. This speeds up wait_us in particular.', 'value': False}'
DEBUG: Extracting config setting 'custom-tickers'='{'help': 'Support custom tickers in addition to USTICKER and LPTICKER. Turning this off can permit some space and speed optimisations, if characteristics of USTICKER and LPTICKER are known at compile time.', 'value': True}'
DEBUG: Extracting config setting 'xip-enable'='{'help': 'Enable Execute In Place (XIP) on this target. Value is only significant if the board has executable external storage such as QSPIF. If this is enabled, customize the linker file to choose what text segments are placed on external storage', 'value': False}'
DEBUG: Extracting config setting 'default-adc-vref'='{'help': 'Default reference voltage for ADC (float)', 'value': 'NAN'}'
DEBUG: Extracting config setting 'internal-flash-uniform-sectors'='{'help': "Target's internal flash has uniform sector sizes", 'value': True}'
DEBUG: Adding config setting: 'target.console-uart'
DEBUG: Adding config setting: 'target.console-uart-flow-control'
DEBUG: Adding config setting: 'target.network-default-interface-type'
DEBUG: Adding config setting: 'target.deep-sleep-latency'
DEBUG: Adding config setting: 'target.app_offset'
DEBUG: Adding config setting: 'target.header_offset'
DEBUG: Adding config setting: 'target.boot-stack-size'
DEBUG: Adding config setting: 'target.mpu-rom-end'
DEBUG: Adding config setting: 'target.default-form-factor'
DEBUG: Adding config setting: 'target.tickless-from-us-ticker'
DEBUG: Adding config setting: 'target.init-us-ticker-at-boot'
DEBUG: Adding config setting: 'target.custom-tickers'
DEBUG: Adding config setting: 'target.xip-enable'
DEBUG: Adding config setting: 'target.default-adc-vref'
DEBUG: Adding config setting: 'target.internal-flash-uniform-sectors'
DEBUG: Applying override 'target.network-default-interface-type'=''ETHERNET''
DEBUG: Loading JSON file /Users/harmut01/Projects/mbed-examples/blinky-baremetal/mbed_app.json
DEBUG: Loading JSON file /Users/harmut01/Projects/mbed-examples/blinky-baremetal/mbed-os/events/mbed_lib.json
DEBUG: Loading JSON file /Users/harmut01/Projects/mbed-examples/blinky-baremetal/mbed-os/storage/kvstore/direct_access_devicekey/mbed_lib.json
DEBUG: Loading JSON file /Users/harmut01/Projects/mbed-examples/blinky-baremetal/mbed-os/connectivity/drivers/emac/TARGET_RENESAS_EMAC/TARGET_RZ_A1XX/mbed_lib.json
DEBUG: Loading JSON file /Users/harmut01/Projects/mbed-examples/blinky-baremetal/mbed-os/storage/blockdevice/mbed_lib.json
DEBUG: Loading JSON file /Users/harmut01/Projects/mbed-examples/blinky-baremetal/mbed-os/connectivity/cellular/mbed_lib.json
DEBUG: Loading JSON file /Users/harmut01/Projects/mbed-examples/blinky-baremetal/mbed-os/storage/kvstore/kv_config/tdb_internal/mbed_lib.json
DEBUG: Loading JSON file /Users/harmut01/Projects/mbed-examples/blinky-baremetal/mbed-os/drivers/device_key/mbed_lib.json
[... cut ...]
DEBUG: Adding config setting: 'mbed-trace.allocator'
DEBUG: Adding config setting: 'mbed-trace.deallocator'
DEBUG: Applying override 'target.c_lib'=''small''
DEBUG: Applying override 'target.printf_lib'=''minimal-printf''
DEBUG: Applying override 'platform.minimal-printf-enable-floating-point'='False'
DEBUG: Applying override 'platform.stdio-minimal-console-only'='True'
mbed_config.cmake has been generated and written to '/Users/harmut01/Projects/mbed-examples/blinky-baremetal/cmake_build/RZ_A1H/develop/ARM/mbed_config.cmake'
```

rwalton-arm, 2021-03-30:
Thanks for clarifying that it's only with the baremetal example.

I think I understand what the problem is now. The issue is because "baremetal" uses the "requires" config field to specify which components it includes in the config. When "requires" is specified then we only pull in config from the mbed components listed in "requires" (and components listed in the required component's "requires" and so on), we skip everything else.

mbed-tools knows nothing about CMake targets or linked libraries, it only knows how to process the mbed JSON config. This means the library dependency probably also has to be declared in mbed_app.json as described in the [docs.](https://os.mbed.com/docs/mbed-os/v6.9/bare-metal/using-the-bare-metal-profile.html) This isn't ideal as we're duplicating information, but until we rework how the config macros are generated and move that to CMake, it's an accepted nuisance.

If you have added events to the requires field and it still doesn't work, then we have a bug to fix.

harmut01, 2021-03-30:
Thanks @rwalton-arm that fixes it!

rwalton-arm, 2021-04-01:
> Thanks @rwalton-arm that fixes it!

That's good. We still need to figure out how to avoid duplicating this information, as currently the UX is really poor. I suggest we keep this ticket open to track the issue.