ARMmbed/mbed-tools #255: Support generating System Workbench for STM32 (SW4STM32) projects

Patater, 2021-04-01:
**Is your feature request related to a problem? Please describe.**
I'd like to use the [System Workbench for STM32 (SW4STM32) IDE](https://www.st.com/en/development-tools/sw4stm32.html) with my Mbed project. Mbed CLI 1 could export to an SW4STM32 IDE project, but Mbed CLI 2 doesn't provide this ability.

**Describe the solution you'd like**
I'd like to use CMake to generate an SW4STM32 IDE project.

**Describe alternatives you've considered**
None

**Additional context**
None

jeromecoutant, 2021-04-01:
Hi
Note that System Workbench for STM32 is now deprecated,
and replaced by STM32 CubeIDE

https://www.st.com/en/development-tools/stm32cubeide.html

Thx

Patater, 2021-04-01:
@jeromecoutant Thanks! I'll close this issue as we probably don't want to spend time adding support for deprecated IDEs.

I've raised https://github.com/ARMmbed/mbed-tools/issues/257 for STM32CubeIDE.

closed 2021-04-01 by Patater