ARMmbed/mbed-tools #258: This introduces a big issue:

Patater, 2021-04-01:
This introduces a big issue:

````
$ pip list | grep mbed-tools
mbed-tools         7.6.0

$ mbedtools detect
Traceback (most recent call last):
  File "xxx\appdata\local\programs\python\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "xxx\appdata\local\programs\python\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "xxx\AppData\Local\Programs\Python\Python37\Scripts\mbedtools.exe\__main__.py", line 7, in <module>
  File "xxx\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "xxx\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "xxx\appdata\local\programs\python\python37\lib\site-packages\mbed_tools\cli\main.py", line 38, in invoke
    super().invoke(context)
  File "xxx\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "xxx\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "xxx\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "xxx\appdata\local\programs\python\python37\lib\site-packages\mbed_tools\cli\list_connected_devices.py", line 29, in list_connected_devices
    connected_devices = get_connected_devices()
  File "xxx\appdata\local\programs\python\python37\lib\site-packages\mbed_tools\devices\devices.py", line 25, in get_connected_devices
    device = Device.from_candidate(candidate_device)
  File "xxx\appdata\local\programs\python\python37\lib\site-packages\mbed_tools\devices\device.py", line 48, in from_candidate
    device_file_info = read_device_files(candidate.mount_points)
  File "xxx\appdata\local\programs\python\python37\lib\site-packages\mbed_tools\devices\_internal\file_parser.py", line 121, in read_device_files
    details_txt_contents = _read_first_details_txt_contents(device_file_paths)
  File "xxx\appdata\local\programs\python\python37\lib\site-packages\mbed_tools\devices\_internal\file_parser.py", line 162, in _read_first_details_txt_contents
    return _read_details_txt(contents)
  File "xxx\appdata\local\programs\python\python37\lib\site-packages\mbed_tools\devices\_internal\file_parser.py", line 180, in _read_details_txt
    key, value = line.split(":", maxsplit=1)
ValueError: not enough values to unpack (expected 2, got 1)

$ mbedls
| platform_name | platform_name_unique | mount_point | serial_port | target_id                | interface_version |
|---------------|----------------------|-------------|-------------|--------------------------|-------------------|
| NUCLEO_F103RB | NUCLEO_F103RB[0]     | D:          | COM15       | 07000221283E5EF4177AC8C9 | V2J37M26          |

````

_Originally posted by @jeromecoutant in https://github.com/ARMmbed/mbed-tools/issues/235#issuecomment-811982174_

Patater, 2021-04-01:
@jeromecoutant Could you please provide the `details.txt` we are failing on? There is a lot of variation in how Mbed devices implement that file and we've obviously not handled all types that are out there in the wild. Thanks!

We'll add a debug log message for this so it's easier to report next time., in addition to fixing this case.

jeromecoutant, 2021-04-01:
As a general comment you should sometimes check that you have the latest version for STLink for your ST boards :-)
https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM#st-link-fw

````
$ cat /d/DETAILS.TXT
Version: V2J37M26
Build:   May 13 2020 17:07:43
````

closed 2021-04-01 by Patater

Patater, 2021-04-01:
The fix for this issue can be validated with [version 7.6.1.dev5](https://pypi.org/project/mbed-tools/7.6.1.dev5/) or any later version.

jeromecoutant, 2021-04-02:
````
$ mbedtools -vv detect
INFO: Using the offline database to identify boards.
Board name     Serial number             Serial port    Mount point(s)    Build target(s)    Interface Version
-------------  ------------------------  -------------  ----------------  -----------------  -------------------
NUCLEO-WB55RG  066dff545057717867195937  COM26          F:                NUCLEO_WB55RG      V2J38M27

````

PR #259 (2021-04-01): devices: Make parser more lenient to unknown layouts
### Description

We can't guarantee details.txt will be in the form "key: value" with a
":" separator on every line. Sometimes they even end with a trailing
newline. Make the code more lenient by using str.partition instead of
str.split, which will return empty strings if the separator isn't found,
rather than raising an exception.

Fixes #258

### Test Coverage

- [x]  This change is covered by existing or additional automated tests.
- [ ]  Manual testing has been performed (and evidence provided) as automated testing was not feasible.
- [  [... cut]
commit a71e0e56 (2021-04-01): devices: Make parser more lenient to unknown layouts

We can't guarantee details.txt will be in the form "key: value" with a
":" separator on every line. Sometimes they even end with a trailing
newline. Make the code more lenient by using str.partition instead of
str.split, which will return empty strings if the separator isn't found,
rather than raising an exception.

Fixes #258