ARMmbed/mbed-tools #260: cmake --parallel

wmmc88, 2021-04-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

wmmc88, 2021-04-05:
This can actually be done by setting the `CMAKE_BUILD_PARALLEL_LEVEL` env variable before calling `mbed-tools`, but i still think it'd benefit from being a command line option

Patater, 2021-04-07:
This seems like an advanced use case. We'd like to keep our CMake wrapper tool to a minimum that allows folks to get up and running with Mbed quickly with a nice experience. I'll add how to do this with the Ninja build generator in our "Build with CMake" documentation.

wmmc88, 2021-04-07:
Since mbed uses CMake now, the CMake way to specify build parallelism is by using `--parallel [<jobs>], -j [<jobs>]` when calling `cmake --build` or by setting the `CMAKE_BUILD_PARALLEL_LEVEL` environment variable. Either of these ways are agnostic to whether cmake is using ninja or make or smth else under the hood.

The former isnt possible when you call `mbed-tools compile` since theres no way to just forward flags to cmake. Personally I think that having a way to just fwd "extra arguments" to cmake doesnt impact the experience (and i could submit a pr to implement that). 

I've confirmed that the latter (ie using the env variable), actually works when set before calling `mbed-tools compile`, so if you're going to call out using Ninja's "-j X" in the docs, it might be better to just say to set `CMAKE_BUILD_PARALLEL_LEVEL` before using `mbed-tools`.

closed 2021-04-19 by donatieng