ARMmbed/mbed-tools #261: Support custom build profiles (with ability to extend existing profiles)

wmmc88, 2021-04-04:
**Is your feature request related to a problem? Please describe.**
According to https://os.mbed.com/docs/mbed-os/v6.9/program-setup/build-profiles-and-rules.html#user-defined-build-profile, we're supposed to be able to define custom build profiles. Right now in mbed-tools, only debug, develop and release are valid and it doesn't look like there's a way to tell `mbed-tools` to use a custom build profile.

**Describe the solution you'd like**
I'd like to be able specify custom user profiles somewhere in my organization's repo and be able to select them. It would also be useful if there were a way to "extend" the default build profiles. Ex. I might want to use something like Release, but add or remove a single flag.

**Describe alternatives you've considered**
N/A

**Additional context**
We're pretty content with the default build profiles, but wanted to enable some extra warnings like Wpedantic, Wvla, Werror, etc and changing the optimization level. I could do this at the CMakelist level, but it really belongs in the build profile jsons.

Patater, 2021-04-07:
We don't support the custom build profiles from Mbed CLI 1 currently. We aren't looking to get feature parity in all cases with the old tool, as before we didn't have CMake available. With Mbed CLI 1, every workflow, however advanced or complicated, had to be covered with our custom Python scripts. Now that we are using CMake, the general approach is to use CMake directly for non-beginner-level workflows given all the power and flexibility it provides, and to optimize Mbed CLI 2 for ease of learning and getting started.

0xc0170, 2021-04-07:
You can create your own toolchain file and set it up - that is basically the toolchain profile. We are aware of some limitations we need to fix - https://github.com/ARMmbed/mbed-os/issues/14446 ). Feel free to send fixes.

> We're pretty content with the default build profiles, but wanted to enable some extra warnings like Wpedantic, Wvla, Werror, etc and changing the optimization level. I could do this at the CMakelist level, but it really belongs in the build profile jsons.

Correct, use CMake instead.

closed 2021-04-08 by Patater