ARMmbed/mbed-tools #263: Wrong CMake build folder specified in mbed-os-5-docs

Patater, 2021-04-06:
> I don't see any documentation updates in this PR. Could you confirm you've looked over the README and other documentation to determine if we need to update anything? For docs, we might need a follow-on PR to [the Mbed CLI 2 docs in mbed-os-5-docs](https://github.com/ARMmbed/mbed-os-5-docs/tree/development/docs/tools/mbed_cli_2), given that the code and docs live in separate repos.

I believe this pr made documentation on [mbed cli 2](https://os.mbed.com/docs/mbed-os/v6.9/build-tools/use.html) incorrect. Specifically the build folder paths (previously just `cmake_build`) in step 2 and 3 of the `Build the project` heading.

_Originally posted by @wmmc88 in https://github.com/ARMmbed/mbed-tools/issues/175#issuecomment-812925754_

wmmc88, 2021-04-06:
> @wmmc88 Well spotted. The instructions for how to build with CMake still work when followed, if you don't already have a `cmake_build` folder. We can update the docs so that when you manually use `cmake` commands, you end up with a compatible directory layout.

Executing the two cmake commands will not "still work when followed, if you don't already have a `cmake_build` folder" because then nothing would have generated the `mbed_config.cmake` configuration. There are no instructions on getting `mbed_config.cmake` without calling mbed-tools

Patater, 2021-04-07:
Yup, I'm wrong. Apologies. It looks like the configure step is missing, which is still needed for now. It would be nice if we had CMake call into a tool to convert all that Mbed json config to `mbed_config.cmake`, rather than needing some pre-CMake step. I'll update the docs to add the configuration step and to match the path `configure` generates (so you don't have to manually copy `mbed_config.cmake` around anywhere).

closed 2021-04-19 by donatieng