ARMmbed/mbed-tools #264: Minor issues about verbose option

jeromecoutant, 2021-04-06:
**Describe the bug**

Verbose option is not accurate

**To Reproduce**

If you go in mbed-os-example-blinky

````
$ mbedtools compile -m NUCLEO_G031K8 -t ARM
xxxx <normal error>

More information may be available by using the command line option '-v'.
````

As indicated in the console, I tried the command line option '-v':
````
````

````
````

````
````

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Version: [e.g. 2000]

**Mbed (please complete the following information):**
 - Device: [e.g. `DISCO_L475VG_IOT01A`]
 - Mbed OS Version: [e.g. 6.8.0]
 - Mbed CLI 2 Version: [e.g. 7.3.1]

**Additional context**
Add any other context about the problem here.

Patater, 2021-04-06:
Yep, agreed this is silly.

dudmuck, 2021-06-19:
I want to see the compiler command line args.  I updated to mbedtools-7.23.0, but i try -vvv, i see a bunch of debug stuff but not gcc command line.  Is it a ninja limitation?

LDong-Arm, 2021-06-21:
> I want to see the compiler command line args. I updated to mbedtools-7.23.0, but i try -vvv, i see a bunch of debug stuff but not gcc command line. Is it a ninja limitation?

The compiler command lines are printed by the `cmake` command, e.g. when you run `cmake --build <build_dir> -v`. We should let mbed-tools pass `-v` to `cmake`.

I personally suggest:
* keeping the existing `mbed-tools -v compile` (`-v` before subcommand) for mbed-tools python self traces.
* adding `mbed-tools compile -v` (`-v` for subcommand, which is missing and thus Issue 1!) to print compiler commands. This should cause `-v` to be passed to `cmake`.