ARMmbed/mbed-tools #272: Target detection not working for ARM_MUSCA_B1

LDong-Arm, 2021-04-22:
**Describe the bug**
`mbed-tools detect` is not working for ARM_MUSCA_B1. Same with the `-f` option of `mbed-tools compile`, so it's a target detection issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect an ARM_MUSCA_B1 board to PC and wait for it to enumerate
2. Run `mbed-tools detect`. The error is
```
ERROR: Could not identify a board with the product code: '5007'.
No connected Mbed devices found.
```

**Expected behavior**
The target ARM_MUSCA_B1 should be recognised. Output from the old tools (`mbed detect`):
```
[mbed] Detected ARM_MUSCA_B1, port /dev/tty.usbmodem143102, mounted /Volumes/MUSCA_B, interface version 0242:
[mbed] Supported toolchains for ARM_MUSCA_B1
| Target       | mbed OS 2 | mbed OS 5 |    ARM    | uARM |  GCC_ARM  | IAR |
|--------------|-----------|-----------|-----------|------|-----------|-----|
| ARM_MUSCA_B1 |     -     | Supported | Supported |  -   | Supported |  -  |
Supported targets: 1
Supported toolchains: 2
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: macOS
 - Version: 11.2.3 (Big Sur)

**Mbed (please complete the following information):**
 - Device: ARM_MUSCA_B1
 - Mbed OS Version: 6.9.0
 - Mbed CLI 2 Version: 7.10.0

**Additional context**
Add any other context about the problem here.

LDong-Arm, 2021-04-22:
ARM_MUSCA_B1 boards are relatively rare and I have one, so I assigned this issue to myself. It's not critical so I may not fix it right away.

Patater, 2021-04-23:
Could this be because `targets/targets.json` for the B1 doesn't have a `"detect_code"` entry? S1 has it as `5009`. I don't see an entry for B1.

Patater, 2021-04-23:
Looks my proposed fix to add `detect_code` to `targets.json` isn't going to work if the tools don't look at the `detect_code`.

LDong-Arm, 2021-04-23:
> Looks my proposed fix to add `detect_code` to `targets.json` isn't going to work if the tools don't look at the `detect_code`.

Thanks for the PR, it's needed anyway.

rwalton-arm, 2021-04-27:
We probably need to update the online database. Looks like we only have an entry for MUSCA_S1.

closed 2021-04-28 by 0xc0170