ARMmbed/mbed-tools #278: [Mbed CLI 2] mbed_rtx.h , device.h , cmsis.h No such file or directory

16L-YT, 2021-05-09:
**Describe the bug**

I am trying to build my project using CLI2. I already have a working project using Mbed Studio. Furthermore, I have a target I added to `targets.json` inherits AMA3B1KK and FAMILY_Apollo3. I already raised a question on the [forum](https://forums.mbed.com/t/mbed-cli-2-no-known-features-for-cxx-compiler-device-h-and-cmsis-h-are-missing/13098), after reading and doing several trials, I think I did all required steps correctly, but I still have missed some header file while compilation as shown in the log below. However, I find that some targets includes a mbed_rtx.h , device.h , cmsis.h in their target folder. Yet, in Ambiq-based targets these are not found and still compilation in Mbed Studio goes correctly. 

```

C:\Users\DELL\Desktop\job\FW\MBED_Cmake\fw-2021-FW>mbed-tools compile -m SIXTEENLAB_DK -t GCC_ARM
Configuring project and generating build system...
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/DELL/Desktop/job/FW/MBED_Cmake/fw-2021-FW/cmake_build/SIXTEENLAB_DK/develop/GCC_ARM
Building Mbed project...
[1/271] Building CXX object CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_rtc_time.cpp.obj
FAILED: CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_rtc_time.cpp.obj
C:\Users\DELL\Desktop\job\tools\xpack-arm-none-eabi-gcc-10.2.1-1.1\bin\arm-none-eabi-g++.exe @CMakeFiles\fw-2021-FW.dir\mbed-os\platform\source\mbed_rtc_time.cpp.obj.rsp -MD -MT CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_rtc_time.cpp.obj -MF CMakeFiles\fw-2021-FW.dir\mbed-os\platform\source\mbed_rtc_time.cpp.obj.d -o CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_rtc_time.cpp.obj -c ../../../../mbed-os/platform/source/mbed_rtc_time.cpp
In file included from ../../../../mbed-os/platform/source/mbed_rtc_time.cpp:17:
../../../../mbed-os/hal/include/hal/rtc_api.h:24:10: fatal error: device.h: No such file or directory
   24 | #include "device.h"
      |          ^~~~~~~~~~
compilation terminated.
[2/271] Building C object CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_power_mgmt.c.obj
FAILED: CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_power_mgmt.c.obj
C:\Users\DELL\Desktop\job\tools\xpack-arm-none-eabi-gcc-10.2.1-1.1\bin\arm-none-eabi-gcc.exe @CMakeFiles\fw-2021-FW.dir\mbed-os\platform\source\mbed_power_mgmt.c.obj.rsp -MD -MT CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_power_mgmt.c.obj -MF CMakeFiles\fw-2021-FW.dir\mbed-os\platform\source\mbed_power_mgmt.c.obj.d -o CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_power_mgmt.c.obj -c ../../../../mbed-os/platform/source/mbed_power_mgmt.c
In file included from ../../../../mbed-os/platform/include/platform/mbed_power_mgmt.h:22,
                 from ../../../../mbed-os/platform/source/mbed_power_mgmt.c:18:
../../../../mbed-os/hal/include/hal/ticker_api.h:25:10: fatal error: device.h: No such file or directory
   25 | #include "device.h"
      |          ^~~~~~~~~~
compilation terminated.
[3/271] Building C object CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_mpu_mgmt.c.obj
FAILED: CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_mpu_mgmt.c.obj
C:\Users\DELL\Desktop\job\tools\xpack-arm-none-eabi-gcc-10.2.1-1.1\bin\arm-none-eabi-gcc.exe @CMakeFiles\fw-2021-FW.dir\mbed-os\platform\source\mbed_mpu_mgmt.c.obj.rsp -MD -MT CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_mpu_mgmt.c.obj -MF CMakeFiles\fw-2021-FW.dir\mbed-os\platform\source\mbed_mpu_mgmt.c.obj.d -o CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_mpu_mgmt.c.obj -c ../../../../mbed-os/platform/source/mbed_mpu_mgmt.c
In file included from ../../../../mbed-os/platform/include/platform/mbed_mpu_mgmt.h:21,
                 from ../../../../mbed-os/platform/source/mbed_mpu_mgmt.c:18:
../../../../mbed-os/hal/include/hal/mpu_api.h:23:10: fatal error: device.h: No such file or directory
   23 | #include "device.h"
      |          ^~~~~~~~~~
compilation terminated.
[4/271] Building CXX object CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_os_timer.cpp.obj
FAILED: CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_os_timer.cpp.obj
C:\Users\DELL\Desktop\job\tools\xpack-arm-none-eabi-gcc-10.2.1-1.1\bin\arm-none-eabi-g++.exe @CMakeFiles\fw-2021-FW.dir\mbed-os\platform\source\mbed_os_timer.cpp.obj.rsp -MD -MT CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_os_timer.cpp.obj -MF CMakeFiles\fw-2021-FW.dir\mbed-os\platform\source\mbed_os_timer.cpp.obj.d -o CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_os_timer.cpp.obj -c ../../../../mbed-os/platform/source/mbed_os_timer.cpp
In file included from ../../../../mbed-os/platform/include/platform/mbed_power_mgmt.h:22,
                 from ../../../../mbed-os/platform/source/mbed_os_timer.cpp:18:
../../../../mbed-os/hal/include/hal/ticker_api.h:25:10: fatal error: device.h: No such file or directory
   25 | #include "device.h"
      |          ^~~~~~~~~~
compilation terminated.
[5/271] Building CXX object CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_retarget.cpp.obj
FAILED: CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_retarget.cpp.obj
C:\Users\DELL\Desktop\job\tools\xpack-arm-none-eabi-gcc-10.2.1-1.1\bin\arm-none-eabi-g++.exe @CMakeFiles\fw-2021-FW.dir\mbed-os\platform\source\mbed_retarget.cpp.obj.rsp -MD -MT CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_retarget.cpp.obj -MF CMakeFiles\fw-2021-FW.dir\mbed-os\platform\source\mbed_retarget.cpp.obj.d -o CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_retarget.cpp.obj -c ../../../../mbed-os/platform/source/mbed_retarget.cpp
In file included from ../../../../mbed-os/platform/include/platform/SingletonPtr.h:24,
                 from ../../../../mbed-os/platform/cxxsupport/./mstd_mutex:43,
                 from ../../../../mbed-os/platform/source/mbed_retarget.cpp:18:
../../../../mbed-os/platform/include/platform/mbed_atomic.h:22:10: fatal error: cmsis.h: No such file or directory
   22 | #include "cmsis.h"
      |          ^~~~~~~~~
compilation terminated.
[6/271] Building CXX object CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_poll.cpp.obj
FAILED: CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_poll.cpp.obj
C:\Users\DELL\Desktop\job\tools\xpack-arm-none-eabi-gcc-10.2.1-1.1\bin\arm-none-eabi-g++.exe @CMakeFiles\fw-2021-FW.dir\mbed-os\platform\source\mbed_poll.cpp.obj.rsp -MD -MT CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_poll.cpp.obj -MF CMakeFiles\fw-2021-FW.dir\mbed-os\platform\source\mbed_poll.cpp.obj.d -o CMakeFiles/fw-2021-FW.dir/mbed-os/platform/source/mbed_poll.cpp.obj -c ../../../../mbed-os/platform/source/mbed_poll.cpp
In file included from ../../../../mbed-os/platform/include/platform/FileHandle.h:25,
                 from ../../../../mbed-os/platform/source/mbed_poll.cpp:18:
../../../../mbed-os/platform/include/platform/platform.h:28:10: fatal error: device.h: No such file or directory
   28 | #include "device.h"
      |          ^~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ERROR: CMake invocation failed!

``` 

**To Reproduce**
Steps to reproduce the behavior:
1. mbed-tools compile -m <ambiq_user_defined> -t GCC_ARM

**Expected behavior**
Compiling successfully. 

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 10

**Mbed (please complete the following information):**
 - Device: AMA3B1KK
 - Mbed OS Version: 6.9.0
 - Mbed CLI 2 Version: 7.12.0

**Additional context**
N/A

boraozgen, 2021-05-10:
I have the same issue. Related to: https://github.com/ARMmbed/mbed-tools/pull/160

0xc0170, 2021-05-10:
In this case, the files are located under `targets/TARGET_Ambiq_Micro/TARGET_Apollo3/device`. Isn't this in the include paths? Also sdk should be added.

This is taken from the CMakeLists.txt located in TARGET_Apollo3 directory:

```
target_include_directories(mbed-apollo3
    INTERFACE
        device
        sdk/CMSIS/AmbiqMicro/Include
        sdk/mcu
        sdk/mcu/apollo3
        sdk/mcu/apollo3/hal
        sdk/mcu/apollo3/regs
)
```

I assume this is not propagated ?

16L-YT, 2021-05-11:
Yes. I check the mentioned CMakeLists.txt, and it indeed has 'device' directory.

0xc0170, 2021-05-11:
it has but the problem could be it is not propagated. Can you look at CMake generated files what paths look like? 

anyway, can anyone provide a branch to test this - reproduce locally the error?

boraozgen, 2021-05-11:
Now that I notice the [custom board guide](https://os.mbed.com/docs/mbed-os/v6.10/porting/porting-custom-boards.html) has been updated, I realized that I might not have done the CMake integration properly. I'll see if I can get it working by following the guide.

boraozgen, 2021-05-11:
I can confirm that by following the guide and integrating into CMake correctly (hint: one must link the custom target to the derived SoC, e.g. `mbed-stm32f412xg`) I managed to resolve the errors.

0xc0170, 2021-05-12:
Thanks @boraozgen for confirming. We will review the reference.

@16L-YT issue resolved as well?

16L-YT, 2021-05-15:
@0xc0170 Yes I managed finally to make it work. I was using a name for the new target library name in the CMakeLists.txt, inside the new target folder, like this one: **mbed-slab-dk**. while the target folder name is **TARGET_SIXTEENLAB_DK**. After checking other target names I found them all are using the  the same target folder name with lower case ( meaning: mbed-sixteenlab-dk), and it worked. I'm not good with CMake to know why it was wrong. 

I vote: to add this as a note the guide. 

Thanks @boraozgen , your comment helped me too much.

16L-YT, 2021-05-15:
@0xc0170 Yes I managed finally to make it work. I was using a name for the new target library name in the CMakeLists.txt, inside the new target folder, like this one: mbed-slab-dk. while the target folder name is TARGET_SIXTEENLAB_DK. After checking other target names I found them all are using the the same target folder name with lower case ( meaning: mbed-sixteenlab-dk), and it worked. I'm not good with CMake to know why it was wrong.

I vote to add this as a note the guide.

Thanks @boraozgen , your comment helped me too much.

closed 2021-05-15 by 16L-YT