ARMmbed/mbed-tools #279: Flash option does not work on NRF52840_DK

noonfom, 2021-05-11:
**Describe the bug**
The flash option does not work on [NRF52840_DK](https://os.mbed.com/platforms/Nordic-nRF52-DK/) as `mbedtools` looks for 
a `bin` and the output extension is `hex` ([mbed-os/targets/targets.json#L5924](https://github.com/ARMmbed/mbed-os/blob/125f58449dec5d138dfe1605661487140256702a/targets/targets.json#L5924)).

**To Reproduce**
Steps to reproduce the behavior:
1. Build an Mbed OS application and append the flash option:

    ```shell
    mbedtools compile -t <toolchain> -m NRF52840_DK --flash
    ```

2.  The resulting error is shown:

    ```shell
    ERROR: Build program file (firmware) not found <app>/cmake_build/NRF52840_DK/develop/<toolchain>/<app>.bin
    ```   

3. The workaround is to create the `bin` from the `elf` using objcopy and rerun the command. Of course, you can also manually copy the `hex` to DAPLINK.  

**Expected behavior**
On NRF52840_DK, look for a hex not a bin. 

**Desktop (please complete the following information):**
 - OS: Ubuntu Linux
 - Version: 20.04

**Mbed (please complete the following information):**
 - Device: NRF52840_DK
 - Mbed OS Version: 6.10.0
 - Mbed CLI 2 Version: 7.10

**Additional context**
None

rwalton-arm, 2021-05-11:
Thanks for the report.

Does it work if you append the `--hex-file` flag? The full command would be: `mbedtools compile -t <toolchain> -m NRF52840_DK --flash --hex-file`

Let us know if it still fails with the `--hex-file` flag added and we can look into it. Thanks!

LDong-Arm, 2021-05-17:
Auto-selection of image format should be implementable, I've created #281 as a reminder.

LDong-Arm, 2021-06-21:
@noonfom We fixed this in #290. If you upgrade mbed-tools to the latest, image format detection becomes automatic and `--hex-file` doesn't exist anymore.

noonfom, 2021-06-21:
Thanks @LDong-Arm. That's a very useful feature.