ARMmbed/mbed-tools #310: Build Mbed application with out-of-tree modules

ATmobica, 2021-08-13:
**Is your feature request related to a problem? Please describe.**
My Mbed project has a specific source tree:

> mbed_app/
> ├─ mbed_app.json
> third_party/
> ├─ wifi_lib/
> │  ├─ mbed_lib.json

I want to build a Mbed application with out-of-tree WiFi module (Mbed OS compatible [wifi-ism43362](https://github.com/ARMmbed/wifi-ism43362) ) but I do not see the possibility of indicating the path to it and join the configuration from mbed_lib.json.
Now, the only option is to move wifi_lib to mbed_project directory (create a symlink for example) and I would like to avoid it.

**Describe the solution you'd like**
Added option to include .json out-of-tree configuration file when generating mbed config.cmake. For example, this could be a new argument to the _mbed-tools configure_ command.

LDong-Arm, 2021-08-13:
Hi @ATmobica, building out-of-tree modules should be possible. This is controlled by CMake instead of mbed-tools.

Could you try adding the following to mbed_app/CMakeLists.txt:
```cmake
add_subdirectory(${CMAKE_CURRENT_SOURCE_DIR}/../third_party/wifi_lib wifi_lib_build)
```
? Note that the second argument of `add_subdirectory()` (in our case `wifi_lib_build`) is needed when including an out-of-tree module, but it can be named anything.

ATmobica, 2021-08-13:
@LDong-Arm Yes, the building works but the propagation settings from _mbed_lib.json_ file does not.
To build wifi_lib I need some compilation flags which should be created by 

 `mbed-tools configure ` in mbed_config.cmake file.

But I couldn't pass the path to _mbed_lib.json_ file to it.

LDong-Arm, 2021-08-13:
> @LDong-Arm Yes, the building works but the propagation settings from _mbed_lib.json_ file does not.

This is a good point.

The idea behind automatically scanning `mbed_lib.json` under the current project comes from the old Mbed CLI tools (i.e. `mbed compile`, before the time of `mbed-tools compile`) where configuration files are automatically scanned. Whenever scanning is involved, it's hard to look outside the current source tree. For now we have `mbed_lib.json` (and thus, scanning) for backward compatibility with the old tools, so such limitation exists.

> **Describe the solution you'd like**
> Added option to include .json out-of-tree configuration file when generating mbed config.cmake. For example, this could be a new argument to the _mbed-tools configure_ command.

@ARMmbed/mbed-os-core What do you think of @ATmobica's suggestion?

ATmobica, 2021-09-09:
@LDong-Arm Hi, is there any progress or decisions on this issue?