ARMmbed/mbed-tools #311: compile fails, pthreads missing?

tdjastrzebski, 2021-08-28:
Today I attempted to use new mbed-tool for the first time ever. I tried the most trivial test I could think of.

```
mbed-tools new .
mbed-tools compile --toolchain GCC_ARM --mbed-target K64F
```

It failed, probably because `pthreads.h` is nowhere to be found but I am not sure. Below is complete log.

```
$ mbed-tools compile --toolchain GCC_ARM --mbed-target K64F
Configuring project and generating build system...
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Python3: C:/Program Files/Python36/python.exe (found version "3.6.8") found components: Interpreter
-- Checking for Python package prettytable -- found
-- Checking for Python package future -- found
-- Checking for Python package jinja2 -- found
-- Checking for Python package intelhex -- found
-- Found Python: C:/Program Files/Python36/python.exe (found version "3.6.8") found components: Interpreter
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - no
-- Could NOT find Threads (missing: Threads_FOUND)
-- Could NOT find Threads (missing: Threads_FOUND)
CMake Error at CMakeLists.txt:18 (mbed_configure_app_target):
  mbed_configure_app_target Function invoked with incorrect arguments for
  function named: mbed_configure_app_target

CMake Error at CMakeLists.txt:20 (mbed_set_mbed_target_linker_script):
  Unknown CMake command "mbed_set_mbed_target_linker_script".

-- Configuring incomplete, errors occurred!
See also "C:/Temp/mb3/cmake_build/CMakeFiles/CMakeOutput.log".
See also "C:/Temp/mb3/cmake_build/CMakeFiles/CMakeError.log".
ERROR: CMake invocation failed!

More information may be available by using the command line option '-v'.
```

mbed-tools is the latest version obtained with: `pip install mbed-tools`. It introduces itself as version **5.0.0** - which may be a bug.

ladislas, 2021-08-28:
the issue is not with pthread, the error is cmake related, as written in your message:

```
CMake Error at CMakeLists.txt:20 (mbed_set_mbed_target_linker_script):
  Unknown CMake command "mbed_set_mbed_target_linker_script".
```

`mbed_set_mbed_target_linker_script` was removed in commit 0f733baeae5a8d4cd0214835d2520db38c5988ff in February.

tdjastrzebski, 2021-08-28:
Thank you. Does it mean this tool is currently broken?

ladislas, 2021-08-28:
No it means the example CMakeLists.txt you're using is out of date.

Try the blinky example: https://github.com/ARMmbed/mbed-os-example-blinky

tdjastrzebski, 2021-08-28:
Well, this example has been created by this tool, it contains nothing but simple main.cpp and CMakeLists.txt files.
```
/* mbed Microcontroller Library
 * Copyright (c) 2021 ARM Limited
 * SPDX-License-Identifier: Apache-2.0
 */

#include "mbed.h"

int main()
{
    printf("Hello, Mbed!\n");
    return 0;
}
```
```
# Copyright (c) 2021 ARM Limited. All rights reserved.
# SPDX-License-Identifier: Apache-2.0

cmake_minimum_required(VERSION 3.19.0)

set(MBED_PATH ${CMAKE_CURRENT_SOURCE_DIR}/mbed-os CACHE INTERNAL "")
set(MBED_CONFIG_PATH ${CMAKE_CURRENT_SOURCE_DIR}/.mbedbuild CACHE INTERNAL "")
set(APP_TARGET )

include(${MBED_PATH}/tools/cmake/app.cmake)

add_subdirectory(${MBED_PATH})

add_executable(${APP_TARGET}
    main.cpp
)

mbed_configure_app_target(${APP_TARGET})

mbed_set_mbed_target_linker_script(${APP_TARGET})

project(${APP_TARGET})

target_link_libraries(${APP_TARGET} mbed-os)

mbed_set_post_build(${APP_TARGET})

option(VERBOSE_BUILD "Have a verbose build process")
if(VERBOSE_BUILD)
    set(CMAKE_VERBOSE_MAKEFILE ON)
endif()
```

ladislas, 2021-08-28:
You need to compare the two CMakeLists.txt, look at this one: https://github.com/ARMmbed/mbed-os-example-blinky/blob/master/CMakeLists.txt

As you can see yours is using `mbed_set_mbed_target_linker_script` while the blinky example does not.

If there is something wrong with the tool it's with the `new` command templates. Try using the blinky example cmake file.

ladislas, 2021-08-28:
Actually nothing is wrong, I've just tried the tools and it works perfectly. Try running:

```python
python3 -m pip install -U --user mbed-tools
```

with or without `--user` depending on how you installed it.

ladislas, 2021-08-28:
[image]
[image]

tdjastrzebski, 2021-08-28:
I updated mbed-tools, started once again from scratch and this time it worked. I have to say I am puzzled since I already had the latest version 7.32. Definitely CMakeLists.txt was broken.
It does not compile with `"requires": ["bare-metal"]` in mbed_app.json but maybe this option is no longer available. Anyway, it is a different problem. Thank you.

0xc0170, 2021-08-30:
Should this be closed as resolved? I'll close it now

Baremetal is available via `mbed-baremetal` CMake target

closed 2021-08-30 by 0xc0170