ARMmbed/mbed-tools #313: TARGET_NAME global macro is not set

jeromecoutant, 2021-09-01:
**Describe the bug**

With CLI1, TARGET_NAME is set to the target name

CLI2 doesn't set it

**To Reproduce**

See error console print:
https://github.com/ARMmbed/mbed-os/blob/master/platform/source/mbed_error.c#L557-L558

**Expected behavior**

**Screenshots**

**Desktop (please complete the following information):**

**Mbed (please complete the following information):**

**Additional context**

LDong-Arm, 2021-09-01:
@jeromecoutant Good catch.

I had a quick look, CLI 2 defines `-DTARGET_xxx`, but it's only useful if one expects a given target and wants to check if it is the current target. To get a target name without knowing, yes we miss a macro. It should be super quick to fix in CMake so I'll raise a PR to mbed-os later.

LDong-Arm, 2021-09-01:
https://github.com/ARMmbed/mbed-os/pull/15041 created