ARMmbed/mbed-tools #320: mbed-tools compile fails with bare-metal

taunomagnusson, 2021-11-24:
**Describe the bug**
mbed-tools are unable to compile for bare-metal profile.

**To Reproduce**
```
mbed-tools new test1
cd test1
nano mbed_app.json (Replace original content with content from https://os.mbed.com/docs/mbed-os/v6.15/bare-metal/using-the-bare-metal-profile.html)
{
    "requires": ["bare-metal"],
    "target_overrides": {
        "*": {
            "target.c_lib": "small"
        }
    }
}
mbed-tools compile -m K64F -t GCC_ARM
```

**Expected behavior**
Should produce a bare-metal .bin/.elf, but fails with an error in compilation (See screenshot)
Note that compilation with the old CLI1 mbed tools work just fine.
I.e. If you do (in the same directory):
rm mbed_os.lib (You need to remove this or mbed CLI1 will complain)
mbed new . -c
mbed compile -m K64F -t GCC_ARM
=> This will compile succesfully

**Screenshots**
```
~/test1/>mbed-tools compile -m K64F -t GCC_ARM
/usr/lib/python3/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (1.26.7) or chardet (3.0.4) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
Configuring project and generating build system...
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tauno/test1/cmake_build/K64F/develop/GCC_ARM
Building Mbed project...
[1/164] Building C object CMakeFiles/test1.dir/mbed-os/cmsis/device/rtos/source/mbed_rtos_rtx.c.obj
FAILED: CMakeFiles/test1.dir/mbed-os/cmsis/device/rtos/source/mbed_rtos_rtx.c.obj
/usr/local/gcc-arm-none-eabi-10.3-2021.10/bin/arm-none-eabi-gcc -DARM_MATH_CM4 -DCOMPONENT_FLASHIAP=1 -DCOMPONENT_SD=1 -DCPU_MK64FN1M0VMD12 -DDEVICE_ANALOGIN=1 -DDEVICE_ANALOGOUT=1 -DDEVICE_CRC=1 -DDEVICE_EMAC=1 -DDEVICE_FLASH=1 -DDEVICE_I2C=1 -DDEVICE_I2CSLAVE=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_LPTICKER=1 -DDEVICE_PORTIN=1 -DDEVICE_PORTINOUT=1 -DDEVICE_PORTOUT=1 -DDEVICE_PWMOUT=1 -DDEVICE_RESET_REASON=1 -DDEVICE_RTC=1 -DDEVICE_SERIAL=1 -DDEVICE_SERIAL_ASYNCH=1 -DDEVICE_SERIAL_FC=1 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_TRNG=1 -DDEVICE_USBDEVICE=1 -DDEVICE_USTICKER=1 -DDEVICE_WATCHDOG=1 -DFEATURE_PSA=1 -DFSL_RTOS_MBED -DMBED_CONF_DRIVERS_OSPI_CSN=OSPI_FLASH1_CSN -DMBED_CONF_DRIVERS_OSPI_DQS=OSPI_FLASH1_DQS -DMBED_CONF_DRIVERS_OSPI_IO0=OSPI_FLASH1_IO0 -DMBED_CONF_DRIVERS_OSPI_IO1=OSPI_FLASH1_IO1 -DMBED_CONF_DRIVERS_OSPI_IO2=OSPI_FLASH1_IO2 -DMBED_CONF_DRIVERS_OSPI_IO3=OSPI_FLASH1_IO3 -DMBED_CONF_DRIVERS_OSPI_IO4=OSPI_FLASH1_IO4 -DMBED_CONF_DRIVERS_OSPI_IO5=OSPI_FLASH1_IO5 -DMBED_CONF_DRIVERS_OSPI_IO6=OSPI_FLASH1_IO6 -DMBED_CONF_DRIVERS_OSPI_IO7=OSPI_FLASH1_IO7 -DMBED_CONF_DRIVERS_OSPI_SCK=OSPI_FLASH1_SCK -DMBED_CONF_DRIVERS_QSPI_CSN=QSPI_FLASH1_CSN -DMBED_CONF_DRIVERS_QSPI_IO0=QSPI_FLASH1_IO0 -DMBED_CONF_DRIVERS_QSPI_IO1=QSPI_FLASH1_IO1 -DMBED_CONF_DRIVERS_QSPI_IO2=QSPI_FLASH1_IO2 -DMBED_CONF_DRIVERS_QSPI_IO3=QSPI_FLASH1_IO3 -DMBED_CONF_DRIVERS_QSPI_SCK=QSPI_FLASH1_SCK -DMBED_CONF_DRIVERS_UART_SERIAL_RXBUF_SIZE=256 -DMBED_CONF_DRIVERS_UART_SERIAL_TXBUF_SIZE=256 -DMBED_CONF_NANOSTACK_LIBSERVICE_PRESENT=1 -DMBED_CONF_PLATFORM_CALLBACK_COMPARABLE=1 -DMBED_CONF_PLATFORM_CALLBACK_NONTRIVIAL=0 -DMBED_CONF_PLATFORM_CRASH_CAPTURE_ENABLED=1 -DMBED_CONF_PLATFORM_CTHUNK_COUNT_MAX=8 -DMBED_CONF_PLATFORM_DEEPSLEEP_STATS_VERBOSE=1 -DMBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE=9600 -DMBED_CONF_PLATFORM_ERROR_ALL_THREADS_INFO=0 -DMBED_CONF_PLATFORM_ERROR_FILENAME_CAPTURE_ENABLED=0 -DMBED_CONF_PLATFORM_ERROR_HIST_ENABLED=0 -DMBED_CONF_PLATFORM_ERROR_HIST_SIZE=4 -DMBED_CONF_PLATFORM_ERROR_REBOOT_MAX=1 -DMBED_CONF_PLATFORM_FATAL_ERROR_AUTO_REBOOT_ENABLED=1 -DMBED_CONF_PLATFORM_MAX_ERROR_FILENAME_LEN=16 -DMBED_CONF_PLATFORM_MINIMAL_PRINTF_ENABLE_64_BIT=1 -DMBED_CONF_PLATFORM_MINIMAL_PRINTF_ENABLE_FLOATING_POINT=0 -DMBED_CONF_PLATFORM_MINIMAL_PRINTF_SET_FLOATING_POINT_MAX_DECIMALS=6 -DMBED_CONF_PLATFORM_POLL_USE_LOWPOWER_TIMER=0 -DMBED_CONF_PLATFORM_STDIO_BAUD_RATE=9600 -DMBED_CONF_PLATFORM_STDIO_BUFFERED_SERIAL=0 -DMBED_CONF_PLATFORM_STDIO_CONVERT_NEWLINES=1 -DMBED_CONF_PLATFORM_STDIO_CONVERT_TTY_NEWLINES=1 -DMBED_CONF_PLATFORM_STDIO_FLUSH_AT_EXIT=1 -DMBED_CONF_PLATFORM_STDIO_MINIMAL_CONSOLE_ONLY=0 -DMBED_CONF_PLATFORM_USE_MPU=1 -DMBED_CONF_RTOS_API_PRESENT=1 -DMBED_CONF_RTOS_PRESENT=1 -DMBED_CONF_TARGET_BOOT_STACK_SIZE=0x1000 -DMBED_CONF_TARGET_CONSOLE_UART=1 -DMBED_CONF_TARGET_CUSTOM_TICKERS=1 -DMBED_CONF_TARGET_DEEP_SLEEP_LATENCY=0 -DMBED_CONF_TARGET_DEFAULT_ADC_VREF=NAN -DMBED_CONF_TARGET_INIT_US_TICKER_AT_BOOT=0 -DMBED_CONF_TARGET_INTERNAL_FLASH_UNIFORM_SECTORS=1 -DMBED_CONF_TARGET_MPU_ROM_END=0x0fffffff -DMBED_CONF_TARGET_NETWORK_DEFAULT_INTERFACE_TYPE=ETHERNET -DMBED_CONF_TARGET_TICKLESS_FROM_US_TICKER=0 -DMBED_CONF_TARGET_XIP_ENABLE=0 -DMBED_CRC_TABLE_SIZE=16 -DMBED_MINIMAL_PRINTF -DMBED_RTOS_SINGLE_THREAD -DMBED_SPLIT_HEAP -DMBED_STACK_DUMP_ENABLED=0 -DMBED_TICKLESS -DMBED_TRACE_COLOR_THEME=0 -DMBED_TRAP_ERRORS_ENABLED=1 -DMEM_ALLOC=malloc -DMEM_FREE=free -DTARGET_CORTEX -DTARGET_CORTEX_M -DTARGET_FF_ARDUINO_UNO -DTARGET_FRDM -DTARGET_Freescale -DTARGET_Freescale_EMAC -DTARGET_K64F -DTARGET_KPSDK_CODE -DTARGET_KPSDK_MCUS -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DTARGET_LIKE_MBED -DTARGET_M4 -DTARGET_MBED_PSA_SRV -DTARGET_MCUXpresso_MCUS -DTARGET_MCU_K64F -DTARGET_NAME=K64F -DTARGET_PSA_Target -DTARGET_PSA_V7_M -DTARGET_RTOS_M4_M7 -DTARGET_Target -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D__CMSIS_RTOS -D__CORTEX_M4 -D__FPU_PRESENT=1 -D__MBED__=1 -D__NEWLIB_NANO -I/home/tauno/test1/mbed-os/cmsis/CMSIS_5/CMSIS/RTOS2/RTX/Config -I/home/tauno/test1/mbed-os/cmsis/CMSIS_5/CMSIS/RTOS2/RTX/Include -I/home/tauno/test1/mbed-os/cmsis/CMSIS_5/CMSIS/RTOS2/RTX/Include1 -I/home/tauno/test1/mbed-os/cmsis/CMSIS_5/CMSIS/RTOS2/RTX/Source -I/home/tauno/test1/mbed-os/cmsis/CMSIS_5/CMSIS/RTOS2/Include -I/home/tauno/test1/mbed-os/cmsis/device/rtos/include -I/home/tauno/test1/mbed-os -I/home/tauno/test1/mbed-os/cmsis/device/RTE/include -I/home/tauno/test1/mbed-os/cmsis/device/. -I/home/tauno/test1/mbed-os/drivers/. -I/home/tauno/test1/mbed-os/drivers/./include -I/home/tauno/test1/mbed-os/drivers/./include/drivers -I/home/tauno/test1/mbed-os/drivers/./include/drivers/internal -I/home/tauno/test1/mbed-os/hal/usb/include -I/home/tauno/test1/mbed-os/hal/usb/include/usb -I/home/tauno/test1/mbed-os/hal/include -I/home/tauno/test1/mbed-os/hal/include/hal -I/home/tauno/test1/mbed-os/platform/cxxsupport/. -I/home/tauno/test1/mbed-os/platform/mbed-trace/include -I/home/tauno/test1/mbed-os/platform/mbed-trace/include/mbed-trace -I/home/tauno/test1/mbed-os/platform/source/minimal-printf/. -I/home/tauno/test1/mbed-os/platform/source/. -I/home/tauno/test1/mbed-os/platform/include -I/home/tauno/test1/mbed-os/platform/include/platform -I/home/tauno/test1/mbed-os/platform/include/platform/internal -I/home/tauno/test1/mbed-os/rtos/. -I/home/tauno/test1/mbed-os/rtos/./include -I/home/tauno/test1/mbed-os/rtos/./include/rtos -I/home/tauno/test1/mbed-os/rtos/./include/rtos/internal -I/home/tauno/test1/mbed-os/rtos/./source -I/home/tauno/test1/mbed-os/connectivity/libraries/nanostack-libservice/. -I/home/tauno/test1/mbed-os/connectivity/libraries/nanostack-libservice/./mbed-client-libservice -I/home/tauno/test1/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/. -I/home/tauno/test1/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device -I/home/tauno/test1/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/drivers -I/home/tauno/test1/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api -I/home/tauno/test1/mbed-os/targets/TARGET_Freescale/. -I/home/tauno/test1/mbed-os/cmsis/CMSIS_5/CMSIS/TARGET_CORTEX_M/Include -I/home/tauno/test1/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/. -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -fomit-frame-pointer -g3 -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -c -Os -MD -MT CMakeFiles/test1.dir/mbed-os/cmsis/device/rtos/source/mbed_rtos_rtx.c.obj -MF CMakeFiles/test1.dir/mbed-os/cmsis/device/rtos/source/mbed_rtos_rtx.c.obj.d -o CMakeFiles/test1.dir/mbed-os/cmsis/device/rtos/source/mbed_rtos_rtx.c.obj -c /home/tauno/test1/mbed-os/cmsis/device/rtos/source/mbed_rtos_rtx.c
In file included from /home/tauno/test1/mbed-os/cmsis/device/rtos/include/mbed_rtx_storage.h:31,
[... cut ...]
/home/tauno/test1/mbed-os/cmsis/device/rtos/include/mbed_rtx_conf.h:60: warning: "OS_IDLE_THREAD_STACK_SIZE" redefined
   60 | #define OS_IDLE_THREAD_STACK_SIZE   (MBED_CONF_RTOS_IDLE_THREAD_STACK_SIZE + EXTRA_IDLE_STACK + EXTRA_IDLE_STACK_DEBUG)
      |
```

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 10

**Mbed (please complete the following information):**
 - Device: K64F
 - Mbed OS Version: Latest master
 - Mbed CLI 2 Version: 7.44.0 (Latest available with pip)

**Additional context**
Add any other context about the problem here.

0xc0170, 2021-11-24:
>     "requires": ["bare-metal"],

the problem is cli1 uses requires. mbed-tools do not read this at all. You need to link to baremetal CMake target rather.

The error is quite cryptic to understand what is going on. Anyway, requires is for only cli1, you are better with CMake with mbed-tools and select there what your application needs.

taunomagnusson, 2021-11-26:
Ok, but how do you compile for bare-metal with CMake/mbed-tools if you can't use the "requires" request?

0xc0170, 2021-11-26:
https://github.com/ARMmbed/mbed-os-example-blinky/blob/master/CMakeLists.txt#L25 - instead of mbed-os, use mbed-baremetal CMake target

rwalton-arm, 2021-12-01:
mbed-tools will actually read "requires" when it generates the baremetal mbed_config.cmake. "requires" (along with the mbedignore files) limit the mbed_lib.json files from mbed-os that are parsed by the tool. You will also need to link to the mbed-baremetal target in your CMake, however. mbed-tools is only responsible for generating the mbed_config.cmake file to generate configuration macros, and for calling a cmake command to build the project. The libraries to actually build need to be specified in the CMakeList files. Hope this helps.